marq24/ha-evcc #16: Increase precision of the smart cost limit

JonisK, 2024-08-19:
### Description

If the user sets a smart cost limit, evcc will suggest prices with an interval of 0.005€/kWh in most cases. Currently, the integration rounds to cents with the call round(float(value), 2). 
it would be great if you could increase the number of decimals to at least 3 so Home Assistant can track whether evcc wants to charge based on the cost limit. 

I guess you would need to edit the two occurences of round(float(value), 2) in number.py but might overlook something here. 

Btw, thank you very much for creating this integration!

marq24, 2024-08-22:
@JonisK can this issue be closed?

JonisK, 2024-08-22:
Oh yes, absolutely. Thank you for implementing  it so quickly. Didn’t know it was my turn to close :)

closed 2024-08-22 by JonisK