marq24/ha-evcc #17: evcc/site/prioritySoc

swa72, 2024-08-23:
### Description

Would like to read evcc/site/prioritySoc and could not find it in the available data. Corresponds to the attached setting

[image]

swa72, 2024-08-23:
I did find it but it reports to "Unknown" (at least in my case).

marq24, 2024-08-23:
well then a debug log output could help... since that's how it should look like (side by side)

[image]

swa72, 2024-08-24:
I enabled debug mode. Reloaded the integration and a value became available, briefly. Then "Unknown" again. I changed the value in EVCC to 25% (which got propagated), then unknown again.

[image]

```
2024-08-24 10:02:39.610 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all frequent_data from evcc@192.168.178.143:7070
2024-08-24 10:02:39.610 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://192.168.178.143:7070/api/state?jq={loadpoints:.loadpoints,auxPower:.auxPower,batteryMode:.batteryMode,batteryPower:.batteryPower,batterySoc:.batterySoc,gridCurrents:.gridCurrents,gridPower:.gridPower,homePower:.homePower,homePower:.homePower,pvPower:.pvPower,vehicles:.vehicles}
2024-08-24 10:02:39.616 DEBUG (MainThread) [custom_components.evcc_intg] number of fields after query: 42
2024-08-24 10:02:39.617 DEBUG (MainThread) [custom_components.evcc_intg] Finished fetching evcc_intg data in 0.007 seconds (success: True)
2024-08-24 10:02:39.618 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:02:39.618 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:02:39.620 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:02:39.620 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:02:45.399 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all data from evcc@192.168.178.143:7070
2024-08-24 10:02:45.399 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://192.168.178.143:7070/api/state
2024-08-24 10:02:45.479 DEBUG (MainThread) [custom_components.evcc_intg] number of fields after query: 41
2024-08-24 10:02:45.479 DEBUG (MainThread) [custom_components.evcc_intg] Finished fetching evcc_intg data in 0.080 seconds (success: True)
2024-08-24 10:02:45.479 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all data from evcc@192.168.178.143:7070
2024-08-24 10:02:45.479 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://192.168.178.143:7070/api/state
2024-08-24 10:02:45.485 DEBUG (MainThread) [custom_components.evcc_intg.binary_sensor] BINARY_SENSOR async_setup_entry
2024-08-24 10:02:45.490 DEBUG (MainThread) [custom_components.evcc_intg.button] BUTTON async_setup_entry
2024-08-24 10:02:45.494 DEBUG (MainThread) [custom_components.evcc_intg.number] NUMBER async_setup_entry
2024-08-24 10:02:45.497 DEBUG (MainThread) [custom_components.evcc_intg.select] SELECT async_setup_entry
2024-08-24 10:02:45.513 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:02:45.513 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:02:45.514 DEBUG (MainThread) [custom_components.evcc_intg.sensor] SENSOR async_setup_entry
2024-08-24 10:02:45.527 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:02:45.528 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:02:45.528 DEBUG (MainThread) [custom_components.evcc_intg.switch] SWITCH async_setup_entry
2024-08-24 10:03:00.532 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all frequent_data from evcc@192.168.178.143:7070
2024-08-24 10:03:00.532 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://192.168.178.143:7070/api/state?jq={loadpoints:.loadpoints,auxPower:.auxPower,batteryMode:.batteryMode,batteryPower:.batteryPower,batterySoc:.batterySoc,gridCurrents:.gridCurrents,gridPower:.gridPower,homePower:.homePower,homePower:.homePower,pvPower:.pvPower,vehicles:.vehicles}
2024-08-24 10:03:00.532 DEBUG (MainThread) [custom_components.evcc_intg.select] SELECT init is COMPLETED
2024-08-24 10:03:00.535 DEBUG (MainThread) [custom_components.evcc_intg] number of fields after query: 42
2024-08-24 10:03:00.535 DEBUG (MainThread) [custom_components.evcc_intg] Finished fetching evcc_intg data in 0.004 seconds (success: True)
2024-08-24 10:03:00.545 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:03:00.545 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:03:00.546 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:03:00.546 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:03:15.864 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all frequent_data from evcc@192.168.178.143:7070
2024-08-24 10:03:15.865 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://192.168.178.143:7070/api/state?jq={loadpoints:.loadpoints,auxPower:.auxPower,batteryMode:.batteryMode,batteryPower:.batteryPower,batterySoc:.batterySoc,gridCurrents:.gridCurrents,gridPower:.gridPower,homePower:.homePower,homePower:.homePower,pvPower:.pvPower,vehicles:.vehicles}
2024-08-24 10:03:15.874 DEBUG (MainThread) [custom_components.evcc_intg] number of fields after query: 42
[... cut ...]
2024-08-24 10:06:58.779 DEBUG (MainThread) [custom_components.evcc_intg] Finished fetching evcc_intg data in 0.006 seconds (success: True)
2024-08-24 10:06:58.781 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:06:58.781 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:06:58.787 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2
2024-08-24 10:06:58.788 INFO (MainThread) [custom_components.evcc_intg] could not find a connected vehicle at loadpoint: 2

```

marq24, 2024-08-24:
can you please be so kind and eMail (see contact data on my github main page) the output of:

`http://192.168.178.143:7070/api/state`

TIA - the output of `could not find a connected vehicle at loadpoint: 2` is quire irritating for me

marq24, 2024-08-24:
OR - do you have multiple ConfigEntries for the EVCC integration?!

swa72, 2024-08-24:
Nothing to worry about ;-), the second load point is a dehumidifier connected to a Shelly Plug S. See https://github.com/swa72/home-assistant/blob/main/evcc-sanitized.yaml

marq24, 2024-08-24:
in this yaml there is no 'prioritySoc' value provided - so can you pls double check the output of the evcc api states endpoint (http://192.168.178.143:7070/api/state) [if it's not already on the way to my eMail inbox]

marq24, 2024-08-24:
please be so kind and update to the pre-release 2024.8.1 - [make sure that you update to latest HACS to be able to install pre-releases again]...

and please be so kind an provide the complete log output after the reboot - TIA (eMail is again just fine)

marq24, 2024-08-24:
@swa72 it's a bit odd - but still no trace, why the data is not shown up - I have made a new pre release - please update again and share the log output with me - but please be so kind, and let the integration run for 5+ minutes - TIA [even if this will generate quite some output]

swa72, 2024-08-25:
Selecting 90%

Before

[image]

After

[image]

marq24, 2024-08-25:
es geht um das `Starte automatisch wenn über 0%` - diese 0 - die verursacht das Problem [mit dem neuen update dann nicht mehr] - trotzdem interessiert mich, was man machen muss/kann, damit der Wert bei 0% steht...

swa72, 2024-08-25:
That's pretty easy - nothing ;-) I do not recall changing the default value.

marq24, 2024-08-25:
this might be related to "long term evcc users" where the migration path between evcc version is might be sometimes difficult...  When you "would" do a clean/fresh install I would not expect, that the default value will be still '0%'

You can do: 1) install the updated Integration, and confirm that the issue is fixed... then once confirmed, try to adjust this 0% value - here in "my" GUI I just can select `Nur mit PV Überschuss` - or a value between 100% and the configured value of `Batterieunterstütztes Fahrzeugladen - wenn Hausbatterie über XX % ist`...

swa72, 2024-08-25:
I can now change `select.evcc_priority_soc` both in your integration and it gets propagated to EVCC and vice versa in EVCC. back to the integration.

If I select "Nur mit genug PV Überschuss" in the EVCC GUI, it displays "wenn über 0 %.". If I again open the selector in the GUI, "Nur mit genug PV Überschuss" is pre-selected, which I find kinda odd.

closed 2024-08-28 by marq24