marq24/ha-evcc #18: EVCC Status ändert sich in Home Assistant aber HA kann nicht EVCC Status ändern

MrMattn, 2024-08-28:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I have noticed, that when I change the status in EVCC, from out to (for example) PV or Quick, it's change in Home Assistant.
But if I change the status in Home Assistant, nothing happens in EVCC.

Here a video: https://youtu.be/zS9fI792XpY

marq24, 2024-08-28:
Please provide debug log output - could be that write operation from HA to evcc is failing... Therefore I need the logging - when you modify the selection in HA

MrMattn, 2024-08-28:
[Uploading home-assistant_evcc_intg_2024-08-28T15-43-51.712Z.log…]()

hilft dir das weiter?

marq24, 2024-08-28:
"das" was hier bislang in Github angekommen ist hilft mir bislang nicht weiter :-D [da ist zwar ein link in Deinem post - aber irgendwie kein Content]

marq24, 2024-08-28:
at least here local with latest evcc 130.6 and ha 2024.8.3 there is no issue at all... "works fine in my office"

MrMattn, 2024-08-28:
hilft dir der dropbox link weiter?
https://www.dropbox.com/scl/fi/9bi2o8eebjgbebki8fto6/home-assistant_evcc_intg_2024-08-28T15-43-51.712Z.log?rlkey=8sjlyc0gcoua8r2asmcsem2u1&dl=0

hier ist die log Datei.

marq24, 2024-08-28:
the URL you entered in the setup is "invalid" - you have entered a '/' after the port - this is the root cause of your issue

closed 2024-08-28 by marq24

MrMattn, 2024-08-28:
oha and how or where can I change this? :-(

marq24, 2024-08-28:
remove the ConfigEntry and just add it again...

https://github.com/marq24/ha-evcc/issues/12

MrMattn, 2024-08-29:
ok thank you I will do this.
Where do I have to do that?
under Settings Integrations? Or under HACS?

marq24, 2024-08-29:
[image]

MrMattn, 2024-08-29:
it worked. thank you very much!

MrMattn, 2025-01-20:
Hi I need your help again.
The problem is back.
I remembered your tip here and removed EVCC and reinstalled it.
I left out "/" at the end of the URL.
But the problem keeps coming up.
The status changes to "PV" every now and then.

How can I create or find this log file that you need?

marq24, 2025-01-20:
https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md

marq24, 2025-01-20:
but probably either evcc and/or the integration are again not up2date...

MrMattn, 2025-01-20:
I think you're right. I should check that first. EVCC is up to date in HA but not in proxmox in the VM in which EVCC is installed.

If I open EVCC I see the version v0.124.6.

Unfortunately I don't know the command how I can update EVCC in the console in Proxmox