marq24/ha-evcc #19: Add support for Homebattery -> Load From Grid

MarcelMertens, 2024-09-17:
### Description

With the latest release EVCC supports load from grid based on energy prices for the home battery (simmilar to the car battery).
[image]

Any chance to add this?

Kind Regards

### API Keys

```shell
0,12€ -> POST /api/batterygridchargelimit/0.12
-0,03€ -> POST /api/batterygridchargelimit/-0.035
```

marq24, 2024-09-18:
sure - should not be a big deal - thanks for let me know!

marq24, 2024-09-18:
update to https://github.com/marq24/ha-evcc/releases/tag/2024.9.1

marq24, 2024-09-20:
@MarcelMertens can we close this?

MarcelMertens, 2024-09-20:
yes, thank you!

closed 2024-09-20 by marq24

bjornenvall, 2025-01-02:
no, it doesn't work because it has a maximal of:

min: -0.05
max: 0.45
step: 0.005
mode: box
unit_of_measurement: SEK/kWh
icon: mdi:cash-multiple
friendly_name: "Home-Battery: grid charging SEK limit ≤"

how can change this?

bjornenvall, 2025-01-02:
[image]

bjornenvall, 2025-01-02:
i want to raise the max value.

marq24, 2025-01-02:
so if you want to enter values > 45 ct/kWh then `native_max_value=0.45` must be adjusted

bjornenvall, 2025-01-02:
yeah, i understand that, how do i do that? as i wrote i tried it, what should i do after?, because it doesn't change in ha. the sensor keeps it old max value. is it in more files other than  const.py?

marq24, 2025-01-02:
let me try to create/provide a fix (that will use 10x of the default max value, if currency is not EUR)

bjornenvall, 2025-01-02:
Okey. So I cant change the value in the file?
Thanks!

marq24, 2025-01-02:
you can - but you need to restart ha that the change will take effect... (and there are two cost limit's that should be adjusted)... fix will be there in 15sec...

bjornenvall, 2025-01-02:
It works! Thanks!

bjornenvall, 2025-01-02:
Hi again..
Now only missing the way to turn it off. Or as it’s called in evcc, “none”.