marq24/ha-evcc #22: Make reconfiguration possible

4n4n4s, 2024-10-06:
### Description

Hi,
I had to swap my evcc instance to another server. Problem is you can only change the IP in the setup process.

Would be nice if we could add some reconfigure option, so we still can change the IP . I saw it in another project.
https://github.com/WillCodeForCats/solaredge-modbus-multi/pull/664/files
Maybe the reference helps.

Workaround for now is to delete and re-create the integration entry.

Thanks.

marq24, 2024-10-06:
and did "remove" and re-add the integration had any negative side effect? [like lost statistics?]

marq24, 2024-10-08:
I have other integrations that allow the change of the IP in the options - but for now I would flag this as "currently won't do"... the case is really just seldom - and as you said, there is a simple workaround - remove and add the integration again

closed 2024-10-08 by marq24