marq24/ha-evcc #26: Detected blocking call to open with args...

nicx, 2024-10-11:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Ill get a WARNING message inside my HA log which forces me to create a bug report :)

#2024-10-11 11:32:11.570 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open with args ('/config/custom_components/evcc_intg/manifest.json',) inside the event loop by custom integration 'evcc_intg' at custom_components/evcc_intg/__init__.py, line 57: with open(f"{basepath}manifest.json") as f: (offender: /config/custom_components/evcc_intg/__init__.py, line 57: with open(f"{basepath}manifest.json") as f:), please create a bug report at https://github.com/marq24/ha-evcc/issues

### Add your DEBUG log output

```shell
2024-10-11 11:32:11.570 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to open with args ('/config/custom_components/evcc_intg/manifest.json',) inside the event loop by custom integration 'evcc_intg' at custom_components/evcc_intg/__init__.py, line 57: with open(f"{basepath}manifest.json") as f: (offender: /config/custom_components/evcc_intg/__init__.py, line 57: with open(f"{basepath}manifest.json") as f:), please create a bug report at https://github.com/marq24/ha-evcc/issues
For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#open
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 189, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/setup.py", line 165, in async_setup_component
    result = await _async_setup_component(hass, domain, config)
  File "/usr/src/homeassistant/homeassistant/setup.py", line 461, in _async_setup_component
    await asyncio.gather(
  File "/usr/src/homeassistant/homeassistant/setup.py", line 463, in <genexpr>
    create_eager_task(
  File "/usr/src/homeassistant/homeassistant/util/async_.py", line 45, in create_eager_task
    return Task(coro, loop=loop, name=name, eager_start=True)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 742, in async_setup_locked
    await self.async_setup(hass, integration=integration)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 594, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/evcc_intg/__init__.py", line 57, in async_setup_entry
    with open(f"{basepath}manifest.json") as f:
```

marq24, 2024-10-11:
update to the latest version of the integration - the code is gone since June...

closed 2024-10-11 by marq24

nicx, 2024-10-11:
@marq24 with HACS i cannot see a new version. installed 2024.6.0, available 2024.6.0

any idea why?

marq24, 2024-10-11:
upgrade HACS ?! - we are already at 2024.10.1

nicx, 2024-10-11:
HACS is on latest version 2.0.1. 
I will open an issue there. Thanks.

marq24, 2024-10-12:
the first thing I would do is: make use of the 3 dots and use the "update" function... if this not show the new Version (which will be displayed a new area!), then remove the Custom repository and re-add it...

nicx, 2024-10-12:
@marq24 I redownloaded the addon manually and it works. Strange that ha-evcc was the only addon which did not notify any updates for months. all other hacs addons where working as expected. thanks anyway for your hint :)