marq24/ha-evcc #33: Lade- und Restladedauer werden falsch berechnet

d3ni3, 2024-10-25:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Moin,

ich habe das Problem das die Lade- und Restladedauer falsch angezeigt wird. Der Wert wird in Kombination mit evcc 0.131.1 falsch berechnet. Ich habe dazu in der const.py folgende Änderungen (factor auf 60) vorgenommen:

```
ExtSensorEntityDescriptionStub(
        tag=Tag.CHARGEDURATION,
        icon="mdi:clock-digital",
        state_class=SensorStateClass.MEASUREMENT,
        native_unit_of_measurement=UnitOfTime.MINUTES,
        #factor = 60000000000,
        factor = 60,
        device_class=SensorDeviceClass.DURATION,
        suggested_display_precision=0
    ),
    ExtSensorEntityDescriptionStub(
        tag=Tag.CHARGEREMAININGDURATION,
        icon="mdi:clock-digital",
        state_class=SensorStateClass.MEASUREMENT,
        native_unit_of_measurement=UnitOfTime.MINUTES,
        device_class=SensorDeviceClass.DURATION,
        #factor = 60000000000,
        factor = 60,
        suggested_display_precision=0
    ),
```

Da von der evcc Api der Wert in Sekunden geliefert wird, muss der factor auf 60 definiert werden. Klappt so bei mir.

marq24, 2024-10-25:
Danke für den Hinweis - der Fix sieht ein wenig anders aus - aber in der Sache hast Du natürlich recht... 

fixed in https://github.com/marq24/ha-evcc/releases/tag/2024.10.2

Bitte Issue schließen, wenn gelöst - merci

d3ni3, 2024-10-25:
Perfekt, danke. Passt nun!

closed 2024-10-25 by d3ni3