marq24/ha-evcc #43: Remove the cost limit via the integration does not work

Steph-74, 2024-11-03:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Hi Matthias,
I found an issue with cost limit "number.evcc_garage_smart_cost_limit":

**Introduction**
When no limit is set in evcc the field is in state "unkown":
[image]

**Steps to reproduce**
Setting a cost limit via the integration works smoothly. But removing the limit (setting the number field to "unkown" or blank) does not work as HA is expecting a float for number fields.

Best
Stephan

marq24, 2024-11-03:
To my knowledge, there is no way to set the value 'unknown' for a number sensor in HA - that's said, implementing the requested functionality would mean either to introduce a select sensor (combobox)... or an additional button sensor to reset the value to `null`

marq24, 2024-11-03:
here you go -> https://github.com/marq24/ha-evcc/releases/tag/2024.11.0

Steph-74, 2024-11-03:
Thanks, will  have a look after the next update cycle for my HA instance ...

closed 2024-11-11 by marq24