marq24/ha-evcc #44: limit_soc didn't get the right value at connect

Syonis, 2024-11-11:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

If I preselect a value for a detected car, it shows everytime 100% the car is connected. But this didn't effect evcc. Evcc will take the right (preselected) value. If I select a different value it sends and shows the correct ammount.

### Add your DEBUG log output

```shell
Not active
```

marq24, 2024-11-11:
I am sorry - but I don't understand your description of the possible issue - you might like to retry it in German? (and might provide screenshots?)

Syonis, 2024-11-11:
Oh, sehr gut. In Deutsch.
Wenn im EVCC einem Fahrzeug ein bestimmter SOC hinterlegt ist so funktioniert das auch im EVCC, doch im Addon wird immer 100% angezeigt. Verstellen lässt sich der SOC aber egal wo. Dann wird auch der jeweils gewählte SOC angezeigt.

marq24, 2024-11-11:
Aktuelles Fahrzeug SOC Limit:
select.evcc_[LOADPOINT]_min_soc

Loadpoint Default SOC Limit
number.evcc_[LOADPOINT]_limit_soc

Syonis, 2024-11-11:
Ok,
im EVCC ist dem Fahrzeug "ZOE" ein max SOC (80%) hinterlegt.
Wenn das Fahrzeug nun angesteckt wird, erkennt EVCC dass es sich um den ZOE handelt und regelt das number.evcc_garage_limit_soc auf 80% herunter. Im Homeassistant wird das aber nicht angezeigt. Wohl aber im EVCC. 
Wenn nun irgendwo geregelt wird, wird auch alles angezeigt und es gibt keine Abweichungen.
Nur wenn EVCC den SOC selbstständig regelt, wird der Wert im HA nicht angezeigt.

Syonis, 2024-11-11:
select.evcc_garage_min_soc  20%
select.evcc_garage_limit_soc 80%

marq24, 2024-11-11:
ich versuch's nochmal zusammenzufassen... [ich denke ich konnte Dir jetzt zumindest inhaltlich folgen]

Soll Zustand: nach dem Einstecken eines Fahrzeuges, sollte das in EVCC konfigurierte SOC-Limit für das Fahrzeug auch als "aktuelles" SOC-Limit (des Ladepunktes) gesetzt werden.

Oder anders formuliert - der Wert für `select.evcc_[LOADPOINT]_limit_soc` sollte von EVCC beim Einstecken des Fahrzeuges als default-Ladelimit (und damit auch der Wert von `number.evcc_[LOADPOINT]_limit_soc`) gesetzt werden.

**Ich kann das "so hier" nicht bestätigen** - also das EVCC das konfigurierte Fahrzeug SOC-Limit beim Einstecken/Wechsel des Fahrzeuges auch als aktuelles Loadpoint SOC-Limit setzt...

[image]
Die Erwartung ist, wenn man das 'Model 3' verbindet, das Loadpoint SOC-Limit von EVCC auf 75% gesetzt wird)

[image]
Und hier ist die Erwartung das beim Anschluss des 'Golf', das Loadpoint SOC-Limit von EVCC auf 85% gesetzt wird)

Ich kann **leider nicht** bestätigen, dass EVCC das setzen des Fahrzeug-SOC-Limits für das LoadPoint SOC-Limit beim Wechsel der Fahrzeuge vornimmt - in meinem lokalen TestSystem bleibt beim Wechsel der an den LoadPoint anschlossenen Fahrzeuge [Golf nach Model 3 (hin & her)] das Loadpoint limit "fest" [egal ob ich die HA Integration benutze oder nicht, und egal ob ich über EVCC die Fahrzeuge wechsel - oder über HA]... In den Beispiel Screenshots bleibt das LP SOC-Limit immer auf 60%... [Erst wenn ich den Slider in HA oder EVCC bewege, ändert sich der Wert auch in dem anderen System]. - Also aus meiner Sicht **WorksAsDesigned**

Das _kann_ daran liegen, das der Test nicht real ist, also man an einem LP ja keine Fahrzeuge wechseln kann... also immer erst ein disconnect erfolgt... -> das kann ich hier aber aus Mangel an Fahrzeugen nicht lokal testen... Ich bleibe bei der Aussage: **Wenn EVCC das Loadpoint SOC-Limit anpasst, dann wird sich (ja nach eingestelltem HA Integrations-Aktualisierungsintervall) auch der Wert von `number.evcc_[LOADPOINT]_limit_soc` ändern.**

Syonis, 2024-11-11:
Jep, ich habe nicht erkannt, dass:
 [evcc] Garage Ladeplanung: Ankunft Standard Ladelimit 
 [evcc] Garage Standard Ladelimit (SOC)
Zwei unterschiedliche sind. Ich ging davon aus, dass das Standard Ladelimit geändert wird.

Vielen Dank und sorry.

closed 2024-11-11 by marq24