marq24/ha-evcc #47: Evcc does not start

Max236-code, 2024-11-18:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

While I have tried to install this add-on both ways manually and automatically, I don't get evcc to start. 
See picture. I run HomeAssistant via VM on an Synology Diskstation. All other Addons are working fine. 

I checked the installation manual but could not find something specific for this case.

Pleas help!

[image]

### Add your DEBUG log output

```shell
2024-11-18 16:13:02.436 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-18 16:13:02.439 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration evcc_intg which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-18 16:13:02.441 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-18 16:13:56.775 ERROR (SyncWorker_4) [pynetgear.router] Netgear ReadTimeout, service 'DeviceConfig:1', method 'CheckNewFirmware', host 192.168.178.2:443 ssl True
2024-11-18 16:13:59.800 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.gesamterzeug_pv_heute (<class 'homeassistant.components.template.sensor.SensorTemplate'>) is using native unit of measurement 'kWh' which is not a valid unit for the device class ('power') it is using; expected one of ['kW', 'W']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+template%22
2024-11-18 16:14:04.340 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.solax_eps_yield_today (<class 'custom_components.solax_modbus.sensor.SolaXModbusSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2024-11-18 16:15:43.461 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all data from evcc@192.168.178.60:7070
2024-11-18 16:15:43.462 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://192.168.178.60:7070/api/state
2024-11-18 16:15:43.469 ERROR (MainThread) [custom_components.evcc_intg] Exception while test credentials: Cannot connect to host 192.168.178.60:7070 ssl:default [Connect call failed ('192.168.178.60', 7070)]
2024-11-18 16:15:47.718 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all data from evcc@192.168.178.60:7070
2024-11-18 16:15:47.718 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://192.168.178.60:7070/api/state
2024-11-18 16:15:47.721 ERROR (MainThread) [custom_components.evcc_intg] Exception while test credentials: Cannot connect to host 192.168.178.60:7070 ssl:default [Connect call failed ('192.168.178.60', 7070)]
2024-11-18 16:15:52.280 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all data from evcc@192.168.178.60:7070
2024-11-18 16:15:52.280 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://192.168.178.60:7070/api/state
2024-11-18 16:15:52.282 ERROR (MainThread) [custom_components.evcc_intg] Exception while test credentials: Cannot connect to host 192.168.178.60:7070 ssl:default [Connect call failed ('192.168.178.60', 7070)]
```

marq24, 2024-11-18:
how about entering a _valid_ URL? -> include 'http:\\'

marq24, 2024-11-18:
... and BTW - this is NO addon - this is a Integration...

Max236-code, 2024-11-18:
I checked both, the http:\\  -backslash change and the http://192.168.178.60:7070/api/state - but no access possible. Is there any precondition I need to have on the home assist? 
--> you are right, it does is an integration and installed as one.

marq24, 2024-11-18:
From your HA instance (in your case the Synology), it must be possible to request `http://192.168.178.60:7070/api/state` - concerning to your logs this request is failing - might be a firewall issue on the synology - or any other network related thing... you might like to check, if you can use curl from the synology shell ?!

marq24, 2024-11-19:
... in general "computer" running HA (with the installed integration) must have access via http(s) to your "computer" running evcc instance - if computers are virtual (e.g. Docker Containers) you have to solve the issue in your infrastructure setup - nothing that can be solved by any integration implementation.

closed 2024-11-19 by marq24