marq24/ha-evcc #49: HA installation problems

Jambola77, 2024-12-10:
### Checklist

- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I tried to add a new addon repository, but it fails (please see picture)
[image]

### Add your DEBUG log output

```shell
Logfile:
Logger: homeassistant.components.hassio
Quelle: components/hassio/websocket_api.py:135
Integration: Home Assistant Supervisor (Dokumentation, Probleme)
Erstmals aufgetreten: 17:24:17 (2 Vorkommnisse)
Zuletzt protokolliert: 17:27:38
Failed to to call /store/repositories - https://github.com/marq24/ha-evcc is not a valid add-on repository
```

marq24, 2024-12-11:
... well 

this is an **integration** and _not_ an **addon**

closed 2024-12-11 by marq24

Jambola77, 2024-12-11:
Sorry, ich bin schaffe es nicht. Wenn ich eine Github Integration hinzufüge, dann will er sich mit meinem Account verknüpfen. Könntest Du bitte etwas Hilfestellung geben? Besten Dank!

marq24, 2024-12-11:
https://github.com/marq24/ha-evcc?tab=readme-ov-file#installation