marq24/ha-evcc #51: Support grid charging settings

andreas-bulling, 2024-12-15:
### Description

It would be fantastic if the integration allowed to read/set the different grid charging settings, e.g. the "price limit", "Fast charge from home battery" etc

This is currently not possible IMHO

marq24, 2024-12-15:
Mahlzeit! Da es in evcc so viele unterschiedliche Schalter/Optionen gibt, ist es (zumindest für mich) nicht immer ersichtlich von welchen evcc optionen gesprochen wird. Ein Screenshot aus evcc der das entsprechenden Setting zeigt macht es deutlich einfacher vor verstehen, wovon Du sprichst/was Du meinst... 

Desweiteren sollte man wissen, das nicht alle verfügbaren HA-Entities per Default aktiviert sind und man diese erst noch für die Integration in HA aktivieren muss.

andreas-bulling, 2024-12-15:
Hi, sorry, dachte Grid Charging ist so zentral, dass ein Screenshot nicht notwendig ist. Habe einen ergänzt.

UDicke, 2024-12-15:
> to read/set the different grid charging settings, e.g. the "price limit", "Fast charge from home battery" etc

Hallo @andreas-bulling ,
folgende Entitäten sind von der "evcc" Hausbatterie:
[image]

Zum Setzten vom Preislimit schau mal hier: [https://github.com/marq24/ha-evcc/discussions/24](https://github.com/marq24/ha-evcc/discussions/24)

Alle anderen Entitäten sind "evcc Konfiguration" und nicht zum direkten Steuern geeignet. Preislimit eigentlich auch, das hat aber nach Setzen sofort Wirkung und Netzladen wird aktiviert oder deaktiviert genau wie beim Setzen des Preislimits in evcc.

marq24, 2024-12-15:
Nun - wie @UDicke schon sagt - so was offensichtliches gibt es seit dem ersten Tag - such mal nach `number.????_smart_cost_limit` bzw eben `number.evcc_battery_grid_charge_limit`... 

Keine Ahnung, wieso diese Wert bei Dir ggf nicht auftauchen?!

closed 2024-12-16 by marq24

bjornenvall, 2025-01-02:
no, it doesn't work because  it has a maximal of:

min: -0.05
max: 0.45
step: 0.005
mode: box
unit_of_measurement: SEK/kWh
icon: mdi:cash-multiple
friendly_name: "Home-Battery: grid charging SEK limit ≤"

how can change this?