marq24/ha-evcc #52: Power in Watt statt KW

marcogracklauer, 2024-12-20:
### Description

Ich versuche in Home Assistant einen Sensor zu erstellen, der mir den aktuellen Hausverbrauch ohne Wallbox berechnet. Da ich zwei Wechselrichter habe, die jeweils ihre Daten in Watt veröffentlichen und einen Tibber Pulse, der den Netzbezug ebenfalls in Watt anzeigt, wäre es klasse auch die Wallbox in Watt berechnen zu können. Die Kilowatt einfach mit 1000 zu multiplizieren bringt Rundungsfehler mit sich, da die kW nur zwei Dezimalstellen hatten. In evcc ist der Power-Wert in Watt angegeben. Wäre es möglich diesen auch in HA zu übernehmen?

marq24, 2024-12-20:
von welchem EVCC Sensor sprichst Du - normalerweise kann man die Einheit und die Genauigkeit über die Einstellungen der Entity anpassen...

marcogracklauer, 2024-12-20:
Mir gehts um charge_power. Steht in meinem evcc-Log bspw. als [lp-2 ] DEBUG 2024/12/20 23:20:12 charge power: 0W.
OK, ich wusste nicht, dass ich die Werte darin anpassen kann. Ich dachte, es würden ggf. nur die kW-Werte übernommen (da in der evcc-Oberfläche auch nur die kW stehen). Probiere ich morgen direkt aus :-)

marcogracklauer, 2024-12-21:
Danke, es funktionierte tatsächlich insofern, dass ich die Genauigkeit auf 4 Nachkommastellen angepasst habe -> somit komm ich auch zum Ergebnis und kann weiterhin kW im Dashboard anzeigen 👍

closed 2024-12-21 by marcogracklauer