marq24/ha-evcc #56: Support for getting per-car statistics

andreas-bulling, 2025-01-02:
### Description

EVCC recently sign. improved the user interface for charging sessions. In particular, detailed per-car statistics are available (e.g. energy charged per month). I am using these for my half-yearly energy statistics (solar, heatpump, car etc). 

I'd be fantastic if one didn't have to go through the months manually using the interface but if entities like "charged energy last month $car" were available.

marq24, 2025-01-02:
Hi Andreas,

I can't find any trace of a new `vehicle` related statistic field inside the JSON response from the API backend - this might be related to the fact, that I just have a single car configured? - Could you please share, 'where' the new data is provided via the API? - TIA

andreas-bulling, 2025-01-02:
I don't know if there is a dedicated field but it seems

marq24, 2025-01-02:
after some further research - the '/api/sessions' endpoint can be used filter the events (charging sessions) by vehicle - but this all happens currently in the evcc frontend logic - so if you want this data you might like to do some experiments with a own REST-Sensor to filter for some specific cars... [and while I am typing this - your response just appeared - so "good morning"]

andreas-bulling, 2025-01-02:
I was hoping this could be something that your integration could provide - I could imagine many people will be interested in this but do not know how to implement their own REST-sensor to achieve this (including myself)

marq24, 2025-01-02:
Based on the docs the `api/sessions` does not support any additional query parameter - from inspecting the evcc frontend you can see, that all data is filtered inside the frontent code - the backend just provide the full data set (at least for me - might be related to the fact, that I just started last year)... 

The response object from,`api/sessions` is not very "ha-friendly" - since you get a big chunk of data per session (loading event)... and then you have to parse the strings...

just like this one...

```
    {
      "id": 78,
      "created": "2024-XX-XXT15:30:24.097542829+01:00",
      "finished": "2024-XX-XXT15:33:44.106620252+01:00",
      "loadpoint": "XXXXX",
      "identifier": "",
      "vehicle": "MachE XX-XXXXE",
      "odometer": null,
      "meterStart": 868.867,
      "meterStop": 868.933,
      "chargedEnergy": 0.0660000000000309,
      "chargeDuration": 200000000000,
      "solarPercentage": 0,
      "price": 0.0206434536000097,
      "pricePerKWh": 0.3127796,
      "co2PerKWh": null
    },
``` 
each of the entries has it's own id... So as more sessions exist - as more objects (with different id's) you will get...

HA Sensors are designed in a way to work on changes in the same data objects... So when you again take a look at the `api/states` endpoint you see, that the structure will be always the same, it's just the values of the fields that change over time. And this is exactly what's happening inside a simple HA sensor - HA fetch frequently the data and compare "old" with "new" values - and if this value is changed, HA will record this change.

For the `api/session` this will not work, since the data of a session object will not change - you just will get new session entries with new `id`'s over time.

Of course since it's all just software you can still implement vehicle and loadpoint specific energy statistic sensors... all that's need to be done is to fetch frequently stats data (keep track of the last processed `id` - if this last_id changed, sum-up the energy fields per loadpoint & per vehicle name (but where is the end here - you can do plenty of field combinations - why only sum-up the 'chargedEnergy' ?!) and offer this sum as new data to HA... [but you must also consider HA restarts & fresh installations - so the implementation will become a bit more complicated].

I hope these reason explain a bit, why this is not just a simple "please add another field" request and why I am currently don't put this somewhere near the top of the backlog.

andreas-bulling, 2025-01-02:
Thanks a lot for the detailed explanation - fully understood. I've pinged the EVCC developers on Mastodon, explicitly asking for using statistics data in your integration. Maybe they are interested and find a way to either provide data in a more HA-friendly way or additionally the aggregated data.

marq24, 2025-01-18:
I hope it's OK when we will close this?

closed 2025-01-18 by marq24

andreas-bulling, 2025-01-18:
ok