marq24/ha-evcc #57: Don't add Basement as default area at the start of home assistant

JokerGermany, 2025-01-02:
### Checklist

- [X] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I only want to have certain sensors and therefore i only added for certain sensors an area and don't added an error to evcc☀️🚘- Solar Charging.

Would it be possible to not add the area Basement at every start of HomeAssistant?

marq24, 2025-01-02:
well - IMHO (but I can't remember) during the initial configuration of the integration you specify the Area - and this is the one that will be used (also after a restart) - I am not aware, that this can be changed afterwards?!

JokerGermany, 2025-01-03:
I don't think I choosed any area at initial configuration, i think i just deleted basement and clicked on continue.
After every restart i have to change it here:
[image]

marq24, 2025-01-03:
If you have cancelled the initial area setup, then still 'basement' will be the default - the easiest now will be to remove the integration configuration - and create it again [to my experience all previous sensor values will be restored - so technically you will not loose any historic data]...

marq24, 2025-01-03:
mmmmh - looks like that I have hardcoded the 'area' (sorry - just realized it) - I must check this after the Dart-WM-Final...

marq24, 2025-01-03:
updating to https://github.com/marq24/ha-evcc/releases/tag/2025.1.1 should solve your issue - thanks for your report.

closed 2025-01-11 by marq24