marq24/ha-evcc #58: Vehicle name is not resolved

abard35, 2025-01-02:
Very nice tool - thank you very much for it

What am I doing wrong?
In my installation it doesn't resolve the name. I've already seen that it works in the screenshots and also on YouTube. I'm assuming it's more a case of me using it incorrectly than a 'feature'.

[image]
[image]

marq24, 2025-01-02:
where are this `db` entries coming from? I am wondering since in the MQTT explorer (no clue why you share this here) I see, that we have `db:xx` I would expect here a `=` ?! - so I guess something in your evcc.yaml is not 100% correct - you might like to share the output of your http://[IP]:7070/api/state ?!

abard35, 2025-01-02:
Thanks for the answer, as I said, I assume the error is on my side.

The screenshot(MQTT-Explorer) shows that this is what it looks like for me in MQTT, it might be helpful to know that.
 I'm not a developer.

{"result":{"auth":{},"battery":[{"power":1917,"soc":28,"controllable":true}],"batteryCapacity":1,"batteryDischargeControl":true,"batteryEnergy":0,"batteryGridChargeActive":false,"batteryMode":"unknown","batteryPower":1917,"batterySoc":28,"bufferSoc":80,"bufferStartSoc":80,"currency":"CHF","fatal":null,"greenShareHome":1,"greenShareLoadpoints":1,"gridConfigured":true,"gridCurrents":[-4.2,-2.244,-2.749],"gridEnergy":33228.738,"gridPower":-2.483,"gridPowers":[-0.7623000000000001,1.9281000000000001,3.0684],"hems":{},"homePower":1914.517,"influx":{"url":"http://localhost:8086","database":"evcc","org":"","user":"hainfluxuser","insecure":false},"interval":30,"loadpoints":[{"batteryBoost":false,"chargeCurrent":0,"chargeCurrents":[0,0,0],"chargeDuration":0,"chargePower":0,"chargeTotalImport":1640.7,"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"chargerPhases1p3p":true,"chargerPhysicalPhases":null,"chargerStatusReason":"unknown","charging":false,"connected":false,"connectedDuration":9223372036,"disableDelay":180,"disableThreshold":0,"effectiveLimitSoc":100,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanId":0,"effectivePlanSoc":0,"effectivePlanTime":null,"effectivePriority":0,"enableDelay":60,"enableThreshold":0,"enabled":false,"limitEnergy":0,"limitSoc":0,"maxCurrent":16,"minCurrent":6,"mode":"pv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":3,"phasesConfigured":0,"phasesEnabled":0,"planActive":false,"planEnergy":0,"planOverrun":0,"planProjectedEnd":null,"planProjectedStart":null,"planTime":null,"priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"smartCostActive":false,"smartCostNextStart":null,"title":"Carport","vehicleClimaterActive":null,"vehicleDetectionActive":true,"vehicleLimitSoc":0,"vehicleName":"","vehicleOdometer":0,"vehicleRange":0,"vehicleSoc":0,"vehicleWelcomeActive":false}],"mqtt":{"broker":"192.168.1.90:1883","topic":"evcc","user":"mqtt-user"},"network":{"schema":"http","host":"evcc.local","port":7070},"prioritySoc":50,"pv":[{"power":-13.899999999999864}],"pvEnergy":0,"pvPower":0,"residualPower":100,"siteTitle":"Home","smartCostType":"priceforecast","sponsor":{"name":"abard35","expiresAt":"2065-02-14T15:00:00Z"},"statistics":{"30d":{"avgCo2":0,"avgPrice":0.2894370966360523,"chargedKWh":163.27100000000019,"solarPercentage":0.31549195643622574},"365d":{"avgCo2":0,"avgPrice":0.2080715927295343,"chargedKWh":1293.0625002200277,"solarPercentage":58.270646305916216},"thisYear":{"avgCo2":0,"avgPrice":0,"chargedKWh":0,"solarPercentage":0},"total":{"avgCo2":0,"avgPrice":0.2080715927295343,"chargedKWh":1293.0625002200277,"solarPercentage":58.270646305916216}},"tariffFeedIn":0.15,"tariffGrid":0.378,"tariffPriceHome":0.15,"tariffPriceLoadpoints":0.15,"vehicles":{"db:14":{"title":"MS wilde Hilde","icon":"car","capacity":100,"features":["CoarseCurrent"],"repeatingPlans":[]},"db:15":{"title":"Gastfahrzeug","icon":"car","capacity":88,"priority":10,"features":["Offline"],"repeatingPlans":[]},"db:2":{"title":"Mustang MachE","icon":"car","capacity":88,"features":["Offline","Retryable"],"repeatingPlans":[]}},"version":"0.132.0"}}

marq24, 2025-01-02:
can you replace the id's in your evcc.yaml for the vehicles? `db:xx` -> `db_xx` ?!

abard35, 2025-01-02:
I have no car inside the evcc.yaml, maybe this ist eh problem?

`# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  

# unique installation id
# plant: 

plant: 

interval: 30s # control cycle interval

# TOKEN
sponsortoken: 

telemetry: true

modbusproxy:
  - port: 502
    uri: 192.168.1.65:1502

meters:
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.1.90  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 10s  
  name: grid1

- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.1.90  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 10s  
  name: pv2

- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.1.90  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 10s  
  name: battery3

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.30.21  
  name: goE-Gemini

loadpoints:
- title: Carport
  charger: goE-Gemini
  mode: off
#  phases: 3
#  mincurrent: 6
#  maxcurrent: 16
#  resetOnDisconnect: false

site:
  title: Home
  meters:
    grid: grid1
    pv: 
    - pv2
    battery: 
    - battery3
  residualPower: 100
  maxGridSupplyWhileBatteryCharging: 0
  
tariffs:
  currency: CHF # (default CHF)
  grid:
    # static grid price with price zones)
    type: fixed
    price: 0.378 # CHF/kWh
    zones:
      - days: Mo-Fr
        hours: 20-0
        price: 0.2740 # CHF/kWh
      - days: Mo-Fr
        hours: 0-7
        price: 0.2740 # CHF/kWh
      - days: Sa,So
        price: 0.2740 # CHF/kWh
  feedin:
  # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.15 # CHF/kWh

# mqtt message broker
mqtt:
  broker: 192.168.1.90:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: mqtt-user
  password: 

influx:
  url: http://localhost:8086
  database: evcc
  user: hainfluxuser
  password: `

abard35, 2025-01-02:
Nevermind - I will find it....
I know witch entry stand for witch car.... :)

Thanks for assist

marq24, 2025-01-02:
... when you have no vehicles in your evcc.yaml - then you have used "just" the GUI to configure the vehicles... Then this might be a more general problem, that in the `state` response the vehicle identifier can contain `:` [and this needs to be handled then correctly be the integration]...

marq24, 2025-01-03:
Good Morning - good news I can simply reproduce the issue here in my local installations - root cause is the 'vehicle configuration via GUI' - so this must be a more common problem (that I should fix inside the integration) - so some work todo for today...

marq24, 2025-01-03:
... well - please restart your HA installation... this fixed the issue here for me... [when adding new vehicles to evcc via the GUI you also must restart HA in order that the code will pick up the new configuration (this only happens on HA restarts)... [restarting the integration is just not enough]

abard35, 2025-01-04:
sorry for late answer
As you write - must be a local issue. Restart doesn't fix the problem.
Nevermind. You can close this threat. I will find it when I have a little more time.

Thanks for your support

marq24, 2025-01-05:
First of all Thank you! - Second - here is what I tested & reproduced locally in my evcc installation;

1. Added a new (dummy) vehicle in evcc via the GUI
2. After a restart of evcc & HA, the new vehicle showed up in HA with just it's internal id `db:1` [so I was happy to reproduce the issue and I wanted to start to debug the issue]
3. Restarted HA again (with just some additional logging code) - and then the new Vehicle Name showed up correctly... 
4. Then I reverted my changes and restarted HA again... and still the correct name was displayed in the combo-box

That's why I started to believe that's just a restart issue... but if this is not the case for you I am again start to scratch my head.... mhhhh

abard35, 2025-01-05:
[image]
I did it the way you described. Unfortunately it still doesn't work for me.
I only had one other DB:XX vehicle.
Like I said, it's not that bad...
It's definitely due to my installation. 

New dummy vehicle created via UI, HA instance completely restarted. 
Secured again and then normal again... same effect.
Don't waste any more time on it.

Best regards
Albert

marq24, 2025-01-05:
Can I ask you for "a last" thing? Can you please enable (just in the GUI) the debug-log of the integration and then restart the integration - this should generate some information like this line:

`.... [custom_components.evcc_intg] read_evcc_config: LPs: 1 VEHs: 1 CT: 'priceforecast' CUR: €`

In this `read_evcc_config` output - what's the number of `VEH's` reported at the restart?

abard35, 2025-01-05:
2025-01-05 12:14:28.730 DEBUG (MainThread) [custom_components.evcc_intg] read_evcc_config: LPs: 1 VEHs: 3 CT: 'priceforecast' CUR: CHF

marq24, 2025-01-05:
Thanks - *sigh* then I really don't understand, why the code can't pick up the names from this VEH object... This object 'should' contain the 'db:nn' as key and provide then the name...

abard35, 2025-01-05:
hee, that are computers and software....I don't wonder anymore...Armadas of developers by Google, MS, SAP....and in every Update are 'features' don't work for everyone....
As long as it work....no problem....

marq24, 2025-01-05:
... can you email me once the output of /api/states (json file) when you have connected a vehicle with one of the wallboxes... I have to admit, that I am still really like, that "software that I build" just works - this become some sort of habit in the past 40 years... :-D

abard35, 2025-01-05:
I'll do it, but today I've already charged it and just replaced the cable. In any case, I still think so this week

marq24, 2025-01-06:
... when I just looked at the right place of my code I can read:

https://github.com/marq24/ha-evcc/blob/b8b5601051055e0dc6b1e35aa98e1cb33e5ab403/custom_components/evcc_intg/select.py#L90

so could be, that this "patched" translation is not going to work for you - what HA version do you use - is this "latest" 2024.12?

abard35, 2025-01-06:
[image]

abard35, 2025-01-06:
Hallo Matthias,

wie gewünscht
---
{"result":{"auth":{},"battery":[{"power":133,"soc":15,"controllable":true}],"batteryCapacity":1,"batteryDischargeControl":true,"batteryEnergy":0,"batteryGridChargeActive":false,"batteryGridChargeLimit":0,"batteryMode":"unknown","batteryPower":133,"batterySoc":15,"bufferSoc":80,"bufferStartSoc":0,"currency":"CHF","fatal":null,"greenShareHome":1,"greenShareLoadpoints":1,"gridConfigured":true,"gridCurrents":[2.218,-2.94,-3.213],"gridEnergy":33415.878,"gridPower":-5.521,"gridPowers":[12.433,-91.8,-128.29],"hems":{},"homePower":805.279,"influx":{"url":"http://localhost:8086","database":"evcc","org":"","user":"hainfluxuser","insecure":false},"interval":30,"loadpoints":[{"batteryBoost":false,"chargeCurrent":0,"chargeCurrents":[0,0,0],"chargeDuration":0,"chargePower":0,"chargeRemainingEnergy":22222.222,"chargeTotalImport":1667.201,"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"chargerPhases1p3p":true,"chargerPhysicalPhases":null,"chargerStatusReason":"unknown","charging":false,"connected":true,"connectedDuration":0,"disableDelay":180,"disableThreshold":0,"effectiveLimitSoc":80,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanId":0,"effectivePlanSoc":0,"effectivePlanTime":null,"effectivePriority":0,"enableDelay":60,"enableThreshold":0,"enabled":false,"limitEnergy":0,"limitSoc":80,"maxCurrent":16,"minCurrent":6,"mode":"pv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":0,"phasesEnabled":1,"planActive":false,"planEnergy":0,"planOverrun":0,"planProjectedEnd":null,"planProjectedStart":null,"planTime":null,"priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"smartCostActive":false,"smartCostNextStart":null,"title":"Carport","vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleLimitSoc":82,"vehicleName":"db:14","vehicleOdometer":8459.962,"vehicleRange":330,"vehicleSoc":60,"vehicleWelcomeActive":false}],"mqtt":{"broker":"192.168.1.90:1883","topic":"evcc","user":"mqtt-user"},"network":{"schema":"http","host":"evcc.local","port":7070},"prioritySoc":50,"pv":[{"power":677.8000000000001}],"pvEnergy":0,"pvPower":677.8,"residualPower":100,"siteTitle":"Home","smartCostType":"priceforecast","sponsor":{"name":"abard35","expiresAt":"2065-02-14T15:00:00Z"},"statistics":{"30d":{"avgCo2":0,"avgPrice":0.2924331970656881,"chargedKWh":136.70499999999993,"solarPercentage":0.37680177915437696},"365d":{"avgCo2":0,"avgPrice":0.2080715927295343,"chargedKWh":1293.0625002200277,"solarPercentage":58.270646305916216},"thisYear":{"avgCo2":0,"avgPrice":0,"chargedKWh":0,"solarPercentage":0},"total":{"avgCo2":0,"avgPrice":0.2080715927295343,"chargedKWh":1293.0625002200277,"solarPercentage":58.270646305916216}},"tariffFeedIn":0.15,"tariffGrid":0.378,"tariffPriceHome":0.15,"tariffPriceLoadpoints":0.15,"vehicles":{"db:14":{"title":"MS wilde Hilde","icon":"car","capacity":100,"features":["CoarseCurrent"],"repeatingPlans":[]},"db:15":{"title":"Gastfahrzeug","icon":"car","capacity":88,"priority":10,"features":["Offline"],"repeatingPlans":[]},"db:2":{"title":"Mustang MachE","icon":"car","capacity":88,"features":["Offline","Retryable"],"repeatingPlans":[]}},"version":"0.132.0"}}

 

---------------

 

HTH

Albert

marq24, 2025-01-07:
es ist mir echt ein Rätzel... zum einen kommt der Text `-keins-` passend für 'null' aus der Translation-Map - aber die anderen Übersetzungswerte (die Fahrzeugnamen) kommen nicht im Ha Frontend an...

abard35, 2025-01-07:
Ja, so sehe ich es auch, egal, ich kann damit leben. Ich denke wir schließen das jetzt hier.
Schön, dass solange dran geblieben bist, aber das ist jetzt nur noch bedingt sinnvoll, weil der Fehler mit Sicherheit irgendwo in den Wirren meiner Installation stecken muss. Sonst wäre das viel früher aufgefallen. Und hier einen 'Workaround' würde ich an Deiner Stelle auch nicht machen wollen....das wäre nur Chaosoptimierung. Offensichtlich funktioniert es ja sonst überall.

closed 2025-01-07 by marq24