marq24/ha-evcc #59: Can we have the maxPower parameter (Load Management) available?

RikMe, 2025-01-02:
### Description

Would be great to be able to dynamically set the maxPower parameter for load management. 

Potential use case: "Capaciteitstarief" in Belgium.

marq24, 2025-01-03:
What's about the `number.evcc_[LOADPOINT]_limit_energy` entity? You can set this to any kWh (or remove it by setting this value to 0) - if this is not the evcc parameter you want to adjust - please be so kind and provide a screenshot of evcc showing the "setting"... TIA

RikMe, 2025-01-03:
Hi, thanks for following up!

Thats not the paramter. I'm using the load management function for the entire circuit (https://docs.evcc.io/en/docs/features/loadmanagement)

Here you can set a fixed 'maxpower' value. However this can be dynamically adjusted trough MQTT. I have it working trough MQTT now, but would be great if this parameter would be available in HA directly.

MQTT workaround:
```
circuits:
  - name: main # eindeutiger Name, wird als Referenz für andere Circuits verwendet
    title: Hoofdzekering # Anzeige in der UI (später)
    maxPower: 4500 # (optional)
    GetMaxPower:
      source: mqtt
      topic: sensor.evcc_maxpower
    meter: grid # optional
```

marq24, 2025-01-03:
Is there a switch/option in the evcc GUI to adjust this value? [I assume not] - if there is a setting, please be so kind and provide a screenshot... 

From the example you just have shared - you 'can' do the same without MQTT (just by calling HA directly) as it's described here: https://github.com/marq24/ha-evcc/blob/main/HA_AS_EVCC_SOURCE.md

RikMe, 2025-01-03:
Nope, not in the GUI unfortunately. The MQTT option works fine for me now (i'm maintaining an up to data MQTT broker anyway) but would be a great addition to your great integration (if it's available already within EVCC off course...)

pergolafabio, 2026-10-31:
Hi @marq24 , since this week we can define maxPower in the UI, no yaml needed anymore in the UI, is it possible to add this entity in your integration now? would be nice

thnx

[image]

[image]

pergolafabio, 2026-10-31:
@RikMe  can you maybe reopen this issue/fer?

marq24, 2026-10-31:
I have opened the issue again for you @pergolafabio ...

But allow me to ask: why you have the need to frequently adjust the configuration of the maximum power of your grid?

pergolafabio, 2026-10-31:
Hi, in Belgium, we have a peak every month, let's say 4000 watt costs 100Eur extra fee, but if you have peak next month like 5000 anyway, then I also want to charge at a max power of 5000 since that's already the peak...

pergolafabio, 2026-10-31:
This peak demand will be also integrated later in evcc, there is an PR for it... But for now we need to work with maxPower

marq24, 2026-10-31:
if this is anyhow just a temp workaround, then I would kindly ask you you could consider to realize this by a REST-TEMPLATE in HA -> making the adjustment request to the evcc config endpoint.

pergolafabio, 2026-10-31:
im not sure how they are going to implement the peak, its still possible we need to adjust maxPower , the question is already asked like 2 years ago, still not implemented .... so thats why i love to see the maxPower   etc in your integration, dont want to use rest commands etc ... its easier in an integration

marq24, 2026-10-31:
Of course it's easier for _you_ when this is part of the integration... Unfortunately Beside your personal convenience I must consider the maintainability of the overall integration - the current integration core has implemented config-read-only access. So what's looks like a small step for you might end in +10h implementing & testing on my side for (currently just two users). And before you mention it - Yes sure AI _can do_ the change in 20 seconds... but I will not merge it.

pergolafabio, 2026-10-31:
ok, i understand :-)

Well , can you leave it open for now anyway, untill we know more about the new PR with peak demand?

pergolafabio, 2026-11-07:
Hey @marq24 * seems the new PR for peak shaving will be dynamic, nor sure if I want to follow that road, maybe I will stick with the maxPower and adjust it manual each month, based on my grid consumption...

So would love to see the load management entities in your integration 

Thnx for looking into it!

marq24, 2026-11-07:
since I am not following any PR in the evcc I have no clue what you want to tell me here - when with a new evcc release new switches/controls/values will be introduced, **then** please file a new issue here (including the info where in he UI and where in the api/states the new value can be found.

pergolafabio, 2026-11-07:
The PR is actually not related, forget about it...
Just want to see the "load managent" entities in your integration, like maxPower, max current ...

This is possible to change in the UI , I can show screens if you want