marq24/ha-evcc #60: Absolte stats of solar and grid power

ppfeiff, 2025-01-03:
### Description

Awesome that we finally have this integration! Thank you very much.

I love that you offer many sensors that provide the statitics from evcc to home assistant like `sensor.evcc_stat_total_solar_percentage`. However, I'm missing the absolte values that were used to calculate that percentage. This information is avaibale in evcc as "Auswertung Ladeenergie"/"Analysis" (see below). 

[image]

Would it be possible to provide these values (1.598 kwh and 682 kwh) as sensors in home assistant? Ideally, for total/this year/last 30 days just like the percentages that you already provide?

Thank you!

marq24, 2025-01-03:
[image]
?!

ppfeiff, 2025-01-04:
Well, this confused me at first, too. `sensor.evcc_total_charged_kwh` gives me the total kwh used by the wallbox which is 2.272 in my case (the sum of 1.598 + 682). The other gives the percentage which is 70,1%. However, contrary to the naming of these sensors, they don't provide the amount of kwh used from the pv and the grid. There is no simple way to get the separate vales which are 1.598 for pv and 682 for grid. These sensors could be named 'sensor.evcc_stats_total_charged_kwh_pv' and 'sensor.evcc_stats_total_charged_kwh_grid' (analogous for year/last 30 days).

With simple math one can calculate these values but this requires two additional yaml sensors in homeassistant. I was wondering whether there is a simple way to retrieve them directly from evcc just as the other stats sensors already there.

marq24, 2025-01-04:
The integration can provide the data available via the evcc API 'as it is' - as described here https://github.com/marq24/ha-evcc/issues/56 evcc FE will calculate information based on this data...

You can always check the `[evcc-ip]:7070/api/state`  json to check yourself, what information is direct accessible... In HA you might like to make use of the helpers (`[ha-ip]:8123/config/helpers`) in order to do some simple math (based on the existing sensor data from the integration) - instead of use sensor templates in the config - what ever is more convenient for you.

ppfeiff, 2025-01-04:
Thanks for hinting the API. It seems that evcc does not provide these values. 

In this case, there is no other way than to computing them by myself. Thank you!

closed 2025-01-04 by ppfeiff