marq24/ha-evcc #62: EvCC 0.132.1

balgerion, 2025-01-07:
### Checklist

- [X] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

It looks like newest evcc update broke something. 
:) I can upload better logs of needed tomorrow morning (only access to phone right now)

### Add your DEBUG log output

```shell
logger:%20homeassistant.config_entries%0ASource%3A%20config_entries.py%3A640%0AFirst%20occurred%3A%2010%3A27%3A16%20PM%20(1%20occurrences)%0ALast%20logged%3A%2010%3A27%3A16%20PM%0A%0AError%20setting%20up%20entry%20evcc%20for%20evcc_intg%0ATraceback%20(most%20recent%20call%20last)%3A%0A%20%20File%20%22/usr/src/homeassistant/homeassistant/config_entries.py%22,%20line%20640,%20in%20__async_setup_with_context%0A%20%20%20%20result%20=%20await%20component.async_setup_entry(hass,%20self)%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0A%20%20File%20%22/config/custom_components/evcc_intg/__init__.py%22,%20line%2064,%20in%20async_setup_entry%0A%20%20%20%20await%20coordinator.read_evcc_config_on_startup()%0A%20%20File%20%22/config/custom_components/evcc_intg/__init__.py%22,%20line%20194,%20in%20read_evcc_config_on_startup%0A%20%20%20%20%22has_phase_auto_option%22:%20a_loadpoint%5B%22chargerPhases1p3p%22%5D,%0A%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20~~~~~~~~~~~%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%5E%0AKeyError:%20'chargerPhases1p3p'%0A
```

walburgf, 2025-01-07:
I have the same issue running evcc 0.132.1 and HASS 2025.1.1

Error setting up entry evcc for evcc_intg
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/evcc_intg/__init__.py", line 64, in async_setup_entry
    await coordinator.read_evcc_config_on_startup()
  File "/config/custom_components/evcc_intg/__init__.py", line 194, in read_evcc_config_on_startup
    "has_phase_auto_option": a_loadpoint["chargerPhases1p3p"],
                             ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
KeyError: 'chargerPhases1p3p'

steps56, 2025-01-08:
Same at my setup. 😪

patrickweppler, 2025-01-08:
Same here:

2025-01-08 05:34:45.849 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry evcc for evcc_intg
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/evcc_intg/__init__.py", line 64, in async_setup_entry
    await coordinator.read_evcc_config_on_startup()
  File "/config/custom_components/evcc_intg/__init__.py", line 193, in read_evcc_config_on_startup
    "has_phase_auto_option": a_loadpoint["chargerPhases1p3p"],
                             ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
KeyError: 'chargerPhases1p3p'
2025-01-08 05:34:51.325 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry DS124 for synology_dsm
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)

HolgerMiara, 2025-01-08:
Same here:
```
Logger: homeassistant.config_entries
Source: config_entries.py:640
First occurred: January 7, 2025 at 22:43:39 (5 occurrences)
Last logged: 07:54:58

Error setting up entry evcc for evcc_intg
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/evcc_intg/__init__.py", line 64, in async_setup_entry
    await coordinator.read_evcc_config_on_startup()
  File "/config/custom_components/evcc_intg/__init__.py", line 193, in read_evcc_config_on_startup
    "has_phase_auto_option": a_loadpoint["chargerPhases1p3p"],
                             ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
KeyError: 'chargerPhases1p3p'
```

marq24, 2025-01-08:
... I will create a new version of the integration today... root cause is a missing attribute in the loadpoint... I have to get in contact with evcc dev's in order to clarify why the property has been removed from the json response.

marq24, 2025-01-08:
well - so this comit in evcc (and 'bad' coding on my side) are the root cause
https://github.com/evcc-io/evcc/commit/65bdb404941483d8340d81c92505741763a4d930

In this change a new property was introduced `chargerPhaseSwitching` and `chargerPhases1p3p`  had been removed...

in order to make "debugging" even harder this change was already rolled back by https://github.com/evcc-io/evcc/commit/e9d67ee0069a3c088ff9ecbb504a3c0108e64d6d - so the current evcc head still contain the `chargerPhases1p3p` - so it took me a while to understand, why "new" evcc build does not work...

closed 2025-01-08 by marq24

DerBaerFreddy, 2025-01-08:
Thanks for the fast Support! It is working again 😊

steps56, 2025-01-08:
Yes, THANKS A LOT 😍

DerBaerFreddy, 2025-02-06:
Is There a problem again? After my update the integration is Not starting anymore.

marq24, 2025-02-06:
HA Core: 2025.2.0
EVCC: 0.133.0
Integration: 2025.2.1

IF you have exceptions/errors during startup - then please share them in a NEW Issue - but only if you are sure that you are using the  versions listed above

steps56, 2025-02-06:
> HA Core: 2025.2.0 EVCC: 0.133.0 Integration: 2025.2.1
> 
> IF you have exceptions/errors during startup - then please share them in a NEW Issue - but only if you are sure that you are using the versions listed above

Hi,
just updates (Backup before naturally) and now using mentioned all newest Versions.
Everything is working nice as before after restart HA.