marq24/ha-evcc #70: Error after upgrade to Home Assistant 2025.1

andreas-bulling, 2025-01-09:
### Checklist

- [X] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [X] I have installed the **latest** release (or BETA) version of the integration and home assistant.
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

Whether it's directly caused by this I don't know but the temporal correlation suggests that the upgrade to HA 2025.1 might be the reason.

I suddenly see the following error in the log - before the setup has worked perfectly for months already. I didn't change anything else that could have caused this error.

```
Logger: homeassistant.config_entries
Source: config_entries.py:640
First occurred: 14:19:37 (1 occurrences)
Last logged: 14:19:37
Error setting up entry evcc for evcc_intg

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/evcc_intg/__init__.py", line 64, in async_setup_entry
    await coordinator.read_evcc_config_on_startup()
  File "/config/custom_components/evcc_intg/__init__.py", line 194, in read_evcc_config_on_startup
    "has_phase_auto_option": a_loadpoint["chargerPhases1p3p"],
                             ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
KeyError: 'chargerPhases1p3p'
```

daemonenstall, 2025-01-09:
I have the same effect and I also think that this has something to do with last evcc update.

daemonenstall, 2025-01-09:
I've just had a quick view into the latest evcc commit and found this:
```
chargerPhaseSwitching: Boolean => chargerPhases1p3p: Boolean
chargerPhaseReading: Boolean => chargerPhysicalPhases: Number
```
https://github.com/evcc-io/evcc/commit/e9d67ee0069a3c088ff9ecbb504a3c0108e64d6d
[image]

marq24, 2025-01-09:
Why you are all wasting time with this? ***big sig***

this issue has been reported and fixed more then 24h ago #62

closed 2025-01-09 by marq24

daemonenstall, 2025-01-09:
Maybe because the issue was open and HASS did not report any update. I  had to force HACS to update the repo information to show me the update.

marq24, 2025-01-09:
I have updated the issue template, including the info how to force HACS to check for updates...

andreas-bulling, 2025-01-09:
I also wasn't notified about any update, sorry (HACS feels rather slow in this respect I have to say)

andreas-bulling, 2025-01-09:
I just forced the update and noticed that I hadn't received any update notification since version 2024.12.0! Is this a bug in HACS or something on my end/setup?