marq24/ha-evcc #71: Version 2024.12.2 does not initialize because of phase detection

bootix, 2025-01-10:
### Checklist

- [X] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [X] My home assistant version is up to date.
- [X] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [X] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

The initegration is not initializing because it's looking for a parameter "chargerPhases1p3p" in the EVCC api. 
I looked up the api and it does not return a parameter like that in my configuration.
The problem might be based that my loadpoint (Go-E Charger v2) does not support phase switching.

For now I removed the line 194 in __init.py__ and it's working again.

For reference, here is an extract of my evcc.yaml:

```
chargers:
- name: wallbox
  type: template
  template: go-e
  host: goecharger.intra  

loadpoints:
- title: Carport
  charger: wallbox
  vehicle: ev1
  mode: now 
```

Also here is what the API returns on loadpoint 0:
```
0	
batteryBoost	false
chargeCurrent	20
chargeCurrents	
0	0
1	0
2	0
chargeDuration	0
chargePower	0
chargeTotalImport	6037
chargeVoltages	
0	218
1	214
2	214
chargedEnergy	0
chargerFeatureHeating	false
chargerFeatureIntegratedDevice	false
chargerIcon	null
chargerPhaseReading	false
chargerPhaseSwitching	false
charging	false
connected	true
connectedDuration	0
disableDelay	180
disableThreshold	0
effectiveLimitSoc	100
effectiveMaxCurrent	20
effectiveMinCurrent	6
effectivePlanId	0
effectivePlanSoc	0
effectivePlanTime	null
effectivePriority	0
enableDelay	60
enableThreshold	0
enabled	true
limitEnergy	0
limitSoc	0
maxCurrent	20
minCurrent	6
mode	"now"
phaseAction	"inactive"
phaseRemaining	0
phasesActive	3
phasesConfigured	3
phasesEnabled	3
planActive	false
planEnergy	0
planOverrun	0
planProjectedEnd	null
planProjectedStart	null
planTime	null
priority	0
pvAction	"inactive"
pvRemaining	0
sessionCo2PerKWh	null
sessionEnergy	0
sessionPrice	null
sessionPricePerKWh	null
sessionSolarPercentage	0
smartCostActive	false
smartCostNextStart	null
title	"Carport"
vehicleClimaterActive	null
vehicleDetectionActive	false
vehicleLimitSoc	80
vehicleName	"ev1"
vehicleOdometer	111111.793
vehicleRange	357
vehicleSoc	79
vehicleWelcomeActive	false
```

### Add your DEBUG log output

```shell
Logger: homeassistant.config_entries
Source: config_entries.py:640
First occurred: 10:09:43 (1 occurrences)
Last logged: 10:09:43
Error setting up entry evcc for evcc_intg

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/evcc_intg/__init__.py", line 64, in async_setup_entry
    await coordinator.read_evcc_config_on_startup()
  File "/config/custom_components/evcc_intg/__init__.py", line 194, in read_evcc_config_on_startup
    "has_phase_auto_option": a_loadpoint["chargerPhases1p3p"],
                             ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^
KeyError: 'chargerPhases1p3p'
```

marq24, 2025-01-10:
seriously?
[image]

#62

closed 2025-01-10 by marq24

bootix, 2025-01-10:
I am recalling my problem report because it's fixed in the latest version.
I was relying on the HA update mechanism and it did not install latest version.

Manually updating to 2025.1.2 fixed the problem and it's working smooth again!

Thanks for your support, marq24