marq24/ha-evcc #73: SMA no SOC loading/unloading in EVCC

walhai1722, 2025-01-10:
### Checklist

- [X] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [X] My home assistant version is up to date.
- [X] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [X] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA

### Add a description

I I have an SMA hybrid inverter and a BYD battery, but I don't get any SOC or charge/discharge value displayed in Evcc

closed 2025-01-10 by marq24

walhai1722, 2025-01-10:
Was working on my old installation

Closed #73 as not planned.

marq24, 2025-01-10:
again - your request/issue looks like an issue for the evcc repo & team -> https://github.com/evcc-io/evcc/issues

walhai1722, 2025-01-10:
schau ich mir an

walhai1722, 2025-01-10:
any idea which on of this 43 issues?