marq24/ha-evcc #77: Cannot set "number.evcc_battery_grid_charge_limit"

wijnands66, 2025-01-11:
### Checklist

- [X] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [X] My home assistant version is up to date.
- [X] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [X] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [X] I confirm, that I did not read any of the previous bulletin-points and just checked them all, since I don't wanted to waste my time with the details like reading or follow any of the existing hints and instructions. Instead, I prefer that the maintainer of this repro should spend time explaining the things to me, that I have previously ignored all the time. It's the maintainers fault, that they provide open source software and willing to offer support for free.

### Add a description

I can not set the price limit for battery charging from grid. If I do so, I get a warning that the value should not be a "NoneType".

I can set the value via MQTT.

### Add your DEBUG log output

```shell
Logger: homeassistant.components.mqtt.select
Quelle: components/mqtt/select.py:125
Integration: MQTT (Dokumentation, Probleme)
Erstmals aufgetreten: 20:20:12 (2 Vorkommnisse)
Zuletzt protokolliert: 20:20:22

Invalid option for select.battery_grid_charge_limit: '0.145' (valid options: [])
Invalid option for select.battery_grid_charge_limit: '0.26' (valid options: [])
```

UDicke, 2025-01-11:
I can change the value of **number**.battery_grid_charge_limit.**value**

see here: https://github.com/marq24/ha-evcc/discussions/24

marq24, 2025-01-11:
[image]

wijnands66, 2025-01-11:
sorry for clicking the last check box and I don't want to waste anybodies time. But still I have this problem just by filling in a value?

[image]

marq24, 2025-01-12:
Thanks for the additional screenshot - this helps a lot to understand the problem

marq24, 2025-01-12:
please update to 2025.1.4 and close this issue when it's solved

wijnands66, 2025-01-13:
OK, thank you for your patience with inexperienced users like me. I have updated to 2025.1.4 and everything works fine!

closed 2025-01-13 by wijnands66