marq24/ha-evcc #78: Sensor sensor.evcc_grid_power  invalid

walburgf, 2025-01-13:
### Checklist

- [X] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [X] My home assistant version is up to date.
- [X] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [X] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [X] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [X] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [X] I confirm, that I did not read any of the previous bulletin-points and just checked them all.
I don't wanted to waste my time with details, I don't read or follow any existing instructions.
Instead, I want that the maintainers of this repro will spend time explaining the world to me - that's their job!.
I live by the motto: Better to ask than to think about it once.
It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

see log output...

[image]

### Add your DEBUG log output

```shell
ValueError: Sensor sensor.evcc_grid_power has device class 'power', state class 'measurement' unit 'W' and suggested precision '2' thus indicating it has a numeric value; however, it has the non-numeric value: '{'currents': [7.44, 8.255, 8.583], 'energy': 12185.255, 'power': 4184.7, 'powers': [1317.1000000000001, 1302.9, 1564.6000000000001]}' (<class 'dict'>)
2025-01-12 23:08:58.299 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 640, in state
    numerical_value = float(value)  # type:ignore[arg-type]
TypeError: float() argument must be a string or a real number, not 'dict'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 267, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 478, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 177, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 642, in state
    raise ValueError(
    ...<5 lines>...
    ) from err
```

marq24, 2025-01-14:
can you a) share the evcc version number you are currently using... b) when you recently updated evcc, did you also restarted your HA?...

and may I asked for debug output of the evcc integration during HA start (since there I expect some logging-details) - TIA

marq24, 2025-01-14:
I would expect some DEBUG lines from the integration during startup....  
like
```
read_evcc_config: LPs ... GAO: True/False
```
and
```
evcc 'grid' data is available as separate object
```
or
```
evcc 'grid' as prefix
```

walburgf, 2025-01-14:
ah, so I'm on the beta channel.... with evcc version 0.132.1 (ae853058)

marq24, 2025-01-14:
ok - thanks for you report - you will get a new release shortly

marq24, 2025-01-14:
-> https://github.com/marq24/ha-evcc/releases/tag/2025.1.5

please close this issue, when the new version will solve your issue - and Thanks for reading :-D

walburgf, 2025-01-14:
the debug output is no longer seen, but the sensor is still unknown

[image]

walburgf, 2025-01-14:
no, they are also unknown

[image]

walburgf, 2025-01-14:
I assume, that you are using the evcc api: <evcc-ip>:7070/api/state

With me, 
grid power = .result.grid.power
grid current[x] = .result.grid.currents[x]

marq24, 2025-01-14:
just another "try" with https://github.com/marq24/ha-evcc/releases/tag/2025.1.6

if this does not work please provide complete DEBUG log output (after HA restart) - but important DEBUG log must be enabled

walburgf, 2025-01-14:
now it's even worse:

Here's the debug-log..... what I'm wondering ist, that version 0.131.6 is detected, because I have a different version installed.

When I'm calling http://evcc-pve.walburg.net:7070/api/state?jq=.version I get "0.132.1 (ae853058)"

-------------------------------------------------------------------
evcc☀️🚘- Solar Charging
This is a custom integration!
If you have any issues with this you need to open an issue here:
https://github.com/marq24/ha-evcc/issues
-------------------------------------------------------------------

2025-01-14 15:51:03.943 DEBUG (MainThread) [custom_components.evcc_intg] starting evcc_intg for: {'host': 'evcc-pve.walburg.net:7070', 'include_evcc': True, 'name': 'evcc', 'scan_interval': 15, 'sw_version': '0.131.6 (0dad8b82)'}
{'host': 'evcc-pve.walburg.net:7070', 'include_evcc': True, 'name': 'evcc', 'scan_interval': 15, 'sw_version': '0.131.6 (0dad8b82)'}
2025-01-14 15:51:03.943 DEBUG (MainThread) [custom_components.evcc_intg] _async_update_data called
2025-01-14 15:51:03.943 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all data from evcc@http://evcc-pve.walburg.net:7070
2025-01-14 15:51:03.943 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://evcc-pve.walburg.net:7070/api/state
2025-01-14 15:51:03.951 DEBUG (MainThread) [custom_components.evcc_intg] number of fields after query: 42
2025-01-14 15:51:03.951 DEBUG (MainThread) [custom_components.evcc_intg] Finished fetching evcc_intg data in 0.008 seconds (success: True)
2025-01-14 15:51:03.951 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all data from evcc@http://evcc-pve.walburg.net:7070
2025-01-14 15:51:03.951 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://evcc-pve.walburg.net:7070/api/state
2025-01-14 15:51:03.952 DEBUG (MainThread) [custom_components.evcc_intg] read_evcc_config: LPs: 2 VEHs: 2 CT: 'priceforecast' CUR: € GAO: True
2025-01-14 15:51:04.448 DEBUG (MainThread) [custom_components.evcc_intg.binary_sensor] BINARY_SENSOR async_setup_entry
2025-01-14 15:51:04.450 DEBUG (MainThread) [custom_components.evcc_intg.button] BUTTON async_setup_entry
2025-01-14 15:51:04.450 DEBUG (MainThread) [custom_components.evcc_intg.number] NUMBER async_setup_entry
2025-01-14 15:51:04.451 DEBUG (MainThread) [custom_components.evcc_intg.select] SELECT async_setup_entry
2025-01-14 15:51:04.453 DEBUG (MainThread) [custom_components.evcc_intg.sensor] SENSOR async_setup_entry
2025-01-14 15:51:04.453 DEBUG (MainThread) [custom_components.evcc_intg.sensor] evcc 'grid' data is available as separate object
2025-01-14 15:51:04.455 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.evcc_grid_power for domain sensor with platform evcc_intg
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 608, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 927, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1384, in add_to_platform_finish
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 520, in state
    value = self.native_value
            ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/evcc_intg/sensor.py", line 98, in native_value
    value = value[array_idx]
            ~~~~~^^^^^^^^^^^
KeyError: <GRID_CONTENT.POWER: 'power'>
2025-01-14 15:51:04.463 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.evcc_grid_currents_0 for domain sensor with platform evcc_intg
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 608, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 927, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1384, in add_to_platform_finish
    self.async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
[... 87 lines cut ...]
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 478, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
[... cut ...]
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 520, in state
    value = self.native_value
            ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/evcc_intg/sensor.py", line 98, in native_value
    value = value[array_idx]
            ~~~~~^^^^^^^^^^^
KeyError: <GRID_CONTENT.POWER: 'power'>

marq24, 2025-01-14:
*sigh* finally - as always the root of the problem is in most cases in front of the monitor/keyboard... 

I am to stuid to use ENUM's in python - finally update to 2025.1.7 :-D

walburgf, 2025-01-14:
Now i's working fine..... thank you!!!

closed 2025-01-14 by walburgf