marq24/ha-evcc #80: missing battery charging price sensor

stefanriegel, 2025-01-18:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [ ] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repro will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

The integration does not provide a Smart Cost Limit for the Hausbatterie. However, Smart Cost Limits for my two Easee Wallboxes are available in the integration.

I do have the Hausbatterie Charge option configured in EVCC: 
[image]

I can see a toggle to prevent discharge: 
[image]

### Add your DEBUG log output

```shell
2025-01-18 13:25:41.786 DEBUG (MainThread) [custom_components.evcc_intg] _async_update_data called
2025-01-18 13:25:41.786 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all frequent_data from evcc@http://192.168.0.3:7070
2025-01-18 13:25:41.786 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://192.168.0.3:7070/api/state?jq={loadpoints:.loadpoints,auxPower:.auxPower,batteryMode:.batteryMode,batteryPower:.batteryPower,batterySoc:.batterySoc,grid:.grid,gridCurrents:.gridCurrents,gridPower:.gridPower,homePower:.homePower,homePower:.homePower,pvEnergy:.pvEnergy,pvPower:.pvPower,pv:.pv,vehicles:.vehicles,statistics:.statistics}
2025-01-18 13:25:41.791 DEBUG (MainThread) [custom_components.evcc_intg] number of fields after query: 47
2025-01-18 13:25:41.792 DEBUG (MainThread) [custom_components.evcc_intg] Finished fetching evcc_intg data in 0.006 seconds (success: True)
2025-01-18 13:25:42.562 DEBUG (MainThread) [custom_components.evcc_intg] clear_data called...
2025-01-18 13:25:42.563 DEBUG (MainThread) [custom_components.evcc_intg] starting evcc_intg for: {'host': 'http://192.168.0.3:7070', 'include_evcc': True, 'name': 'evcc', 'scan_interval': 5, 'sw_version': '0.132.1'}
{'host': 'http://192.168.0.3:7070', 'include_evcc': True, 'name': 'evcc', 'scan_interval': 15, 'sw_version': '0.132.1'}
2025-01-18 13:25:42.564 DEBUG (MainThread) [custom_components.evcc_intg] _async_update_data called
2025-01-18 13:25:42.564 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all data from evcc@http://192.168.0.3:7070
2025-01-18 13:25:42.564 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://192.168.0.3:7070/api/state
2025-01-18 13:25:42.568 DEBUG (MainThread) [custom_components.evcc_intg] number of fields after query: 45
2025-01-18 13:25:42.569 DEBUG (MainThread) [custom_components.evcc_intg] Finished fetching evcc_intg data in 0.005 seconds (success: True)
2025-01-18 13:25:42.569 INFO (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all data from evcc@http://192.168.0.3:7070
2025-01-18 13:25:42.569 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://192.168.0.3:7070/api/state
2025-01-18 13:25:42.589 DEBUG (MainThread) [custom_components.evcc_intg] read_evcc_config: LPs: 2 VEHs: 2 CT: 'priceforecast' CUR: € GAO: False
2025-01-18 13:25:42.590 DEBUG (MainThread) [custom_components.evcc_intg.binary_sensor] BINARY_SENSOR async_setup_entry
2025-01-18 13:25:42.604 DEBUG (MainThread) [custom_components.evcc_intg.button] BUTTON async_setup_entry
2025-01-18 13:25:42.614 DEBUG (MainThread) [custom_components.evcc_intg.number] NUMBER async_setup_entry
2025-01-18 13:25:42.634 DEBUG (MainThread) [custom_components.evcc_intg.select] SELECT async_setup_entry
2025-01-18 13:25:42.657 DEBUG (MainThread) [custom_components.evcc_intg.sensor] SENSOR async_setup_entry
2025-01-18 13:25:42.658 DEBUG (MainThread) [custom_components.evcc_intg.sensor] evcc 'grid' as prefix
2025-01-18 13:25:42.714 DEBUG (MainThread) [custom_components.evcc_intg.switch] SWITCH async_setup_entry
```

marq24, 2025-01-18:
I do have such an entity ;-)

[image]

But it's true, that this entity is not enabled by default - so you must use the Entity settings to enable it... 

Please close, when my assumption is correct - TIA.

stefanriegel, 2025-01-18:
Thank you for the clarification! I’ve already checked the entities, and I don’t see it listed (see screenshot). Could it be that the entity is missing entirely in my setup?

[image]

[image]

i found a error in my logs:

`2025-01-18 13:57:53.968 ERROR (MainThread) [homeassistant.components.number] Error adding entity evcc_intg.evcc_battery_grid_charge_limit for domain number with platform evcc_intg
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 608, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 821, in _async_add_entity
    entry = entity_registry.async_get_or_create(
        self.domain,
    ...<17 lines>...
        unit_of_measurement=entity.unit_of_measurement,
    )
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 858, in async_get_or_create
    _validate_item(
    ~~~~~~~~~~~~~~^
        self.hass,
        ^^^^^^^^^^
    ...<5 lines>...
        unique_id=unique_id,
        ^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 687, in _validate_item
    raise ValueError(
        f"entity_category must be a valid EntityCategory instance, got {entity_category}"
    )
ValueError: entity_category must be a valid EntityCategory instance, got config`

marq24, 2025-01-18:
AUA!

marq24, 2025-01-18:
I feel so bad! - please update to https://github.com/marq24/ha-evcc/releases/tag/2025.1.8

stefanriegel, 2025-01-18:
Awesome!

[image]

closed 2025-01-18 by stefanriegel