marq24/ha-evcc #81: Battery discharge switch in HA uses different control method than manual EVCC lock → no synchronization / not working

stefanriegel, 2025-01-19:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [ ] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repro will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

## Describe the bug
- **When toggling** `switch.evcc_battery_discharge_control` via the ha-evcc integration, the EVCC logs show:
```shell
[site  ] DEBUG 2025/01/19 16:24:39 set battery discharge control: false
[site  ] DEBUG 2025/01/19 16:24:46 set battery discharge control: true
```

- **When manually locking** the battery inside EVCC, the logs show:
```shell
[site  ] DEBUG 2025/01/19 16:26:14 set battery mode: hold
```

- These two methods call different functions within EVCC and don’t synchronize, resulting in inconsistent states between Home Assistant and EVCC.
-- **Additionally**, the lock command coming from the EVCC_intg integration appears to have no effect in EVCC itself.

## Expected Behavior
The battery lock/unlock functionality should use a unified method so that:
1. Toggling the switch in Home Assistant sets the same state as “hold” in EVCC.
2. Manually enabling “hold” in EVCC updates the switch in Home Assistant.

## To Reproduce
1. **Toggle** the `switch.evcc_battery_discharge_control` in Home Assistant.  
2. **Observe** the EVCC logs → `set battery discharge control: false/true`.  
3. **Lock/unlock** the battery in EVCC’s GUI.  
4. **Observe** the EVCC logs → `set battery mode: hold`.  
5. **Notice** that the state in HA does not reflect the new “hold” mode and vice versa.

Batterie Lock executed with a Helper Loadpoint:
[image]

### Add your DEBUG log output

```shell
## EVCC Logs
- **HA Switch Toggle**  

[site  ] DEBUG 2025/01/19 16:24:39 set battery discharge control: false
[site  ] DEBUG 2025/01/19 16:24:46 set battery discharge control: true

- **Manual Lock in EVCC**  
[site  ] DEBUG 2025/01/19 16:26:14 set battery mode: hold
```

stefanriegel, 2025-01-19:
Sometimes you have to restart systems several times and be patient... now it's just working.

marq24, 2025-01-19:
also have in mind, that HA-Integration "poll" the information from evcc based on your configured update-interval... so when you change something in the EVCC-GUI it could take at least the update intervall, till HA will show this change...

EVCC-GUI makes use of a Websocket, so they can instantly update the GUI when data was changed (via external API calls)...

stefanriegel, 2025-01-19:
The issue has reoccurred. I waited for 15 minutes, restarted everything, and then waited another 20 minutes.

The integration switch is toggled to ON, but EVCC is not blocking the battery discharge. However, when I manually set it again via the EVCC load point, it works as expected.

marq24, 2025-01-19:
... what's your integration update interval?

stefanriegel, 2025-01-19:
15 seconds as recommended in the setup process.
EVCC is on 30sec to Huawei

marq24, 2025-01-19:
the HA switch `switch.evcc_battery_discharge_control` will call the API (https://docs.evcc.io/docs/integrations/rest-api#site): `POST /api/batterydischargecontrol/<enable>: battery discharge control (true/false)`

you can check this by calling `http://[your-evcc-ip]:7070/api/state?jq={batteryDischargeControl}`

I have to admit, that I don't see a switch/setting (except from here) - where I would be able to `Lock/unlock the battery in EVCC’s GUI.`

[image]

marq24, 2025-01-19:
might be related?! https://github.com/marq24/ha-evcc/issues/17

marq24, 2025-01-19:
For sure the current implemented switch, is not related to your linked 'lifehack' - I must admit, I haven't understood yet, how this "dummy charger & loadpoint" can provide the expected funtionality... 

At least the second loadpoint should appear (with all entities) in the HA GUI then - but I would expect, that this then stll offers features like Aus/PV/Min/Schnell...

stefanriegel, 2025-01-19:
Sorry, I’ve updated my comment.

Here’s what I’ve found regarding the behavior:
	1.	The toggle works only when a loadpoint is actively “loading.”
	2.	Under these conditions, I can toggle the switch in the integration.
	3.	The battery is locked when the toggle is ON and unlocked when it’s OFF.
	4.	However, the lock is automatically removed as soon as charging stops.

This explains why the toggle behaves as expected when the dummy loadpoint is “loading”—I can toggle it and, after about 15 seconds, the battery locks or unlocks accordingly.

I need to keep this dummy loader because I want to use the integration toggle to prevent battery discharging when the power price is low.

marq24, 2025-01-19:
Ist das denn aber nicht die eigentlich Lösung? [Wenn man das Entladen dieses Loadpointsunterbinden möchte] (aus der LifehackDoku)

> Ebenso kann der Hausspeicher dadurch auch manuell gesperrt werden, indem man den Modus "schnell" auswählt.

Also zum Sperren den Modus dieses "fake" Loadpoints von PV umstellen auf Schnell (dann sollte - laut der Aussage von Andi) die Batterie auch gesperrt sein? 

Wie gesagt  ich kann dem ganzen nicht wirklich folgen, da mein Speicher keine "Entladesperre" hat - und wenn, dann würde ich (immer) versuchen diese Spere & freigabe direkt über die HA-Integration des Speichers zu regeln... [aber wie gesagt, ich habe nur einen eingeschränkt steuerbaren Speicher]

stefanriegel, 2025-01-19:
Ich hab das bis her auch immer aus HASS heraus gemacht, aber da Huawei eine qual in HASS ist wollte ich nur einen der auf den WR rumhünert 😆

Aber das ding kann zu, da es eine
limitation von EVCC.

sorry für den aufwand

closed 2025-01-19 by stefanriegel