marq24/ha-evcc #82: [evcc] Charging plan: Departure charging goal (SOC) is Unknown

timpetersson81, 2025-01-19:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [x] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repro will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

[evcc] Charging plan: Departure charging goal (SOC) is Unkown, even though a plan is set with a charging goal to 30kwh.

[home-assistant_2025-01-19T15-50-52.689Z.log](https://github.com/user-attachments/files/18469392/home-assistant_2025-01-19T15-50-52.689Z.log)

### Add your DEBUG log output

```shell
I attached the log
```

stefanriegel, 2025-01-19:
ausser eine menge Fehler anderer Integrationen sind dort keine Einträge enthalten.
Ich denke du wolltest den Bug bei evcc.io repo auf machen, oder?

marq24, 2025-01-19:
the fact, that all checkmark's have been checked makes this issue invalid...

Setting a kWh target... and expecting that SOC goal will show something?! Might be a Tesla owner? ;-)

closed 2025-01-19 by marq24

timpetersson81, 2025-01-19:
> the fact, that all checkmark's have been checked makes this issue invalid...
> 
> Setting a kWh target... and expecting that SOC goal will show something?! At least we know, where he have parked his car ;-)
> 
> when using ENERGY as loading plan/goal, then the entities `*_plan_energy` and `*_plan_time` must be enabled in the HA integration...

Im not sure what I should use instead, could you point me in the right direction? I don't have any plan_energy og plan_time to work with.

marq24, 2025-01-19:
[image] 

[image]

[image]