marq24/ha-evcc #85: Charging Plan Entities do not show values or update

gorzan14, 2025-01-21:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [ ] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repro will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

The entities that should report charging plan infos (e.g. sensor.evcc_garage_vehicle_plans_soc, sensor.evcc_garage_vehicle_plans_time) have no values an do not update after setting plan. Even if doing this with the action of this evcc-integration.

I have installed evcc with the homeassistant add-on from the add-on store. Version 0.133.0.

Is it possible to fix. I would really like to show the next charging plan time in a HA dashboard.

Log:
Logger: homeassistant.components.sensor
Quelle: components/sensor/__init__.py:547
Integration: Sensor ([Dokumentation](https://www.home-assistant.io/integrations/sensor), [Probleme](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
Erstmals aufgetreten: 23:00:58 (1 Vorkommnisse)
Zuletzt protokolliert: 23:00:58

Entity sensor.evcc_garage_plan_energy (<class 'custom_components.evcc_intg.sensor.EvccSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/marq24/ha-evcc/issues

### Add your DEBUG log output

```shell

```

marq24, 2025-01-22:
Please update to 2025.1.9 and close this issue when it's fixed - thanks for your report

gorzan14, 2025-01-23:
After Update it worked 2 times: The Planed departure time and the planed SoC where updated. The two other entities remained unknown.

In the evening it didn't update again. In evcc a cargingplan for departure is set. See Screenshots.

[image]

[image]

[image]

I would like to see the planed start of carving which os shown in cvcc here:

[image]

Should one of the other entities show this or is this a feature request?

marq24, 2025-01-23:
yes of course - they are about the ENERGY target - that will only work for GUEST vehicles (where there is no option to read SOC)

closed 2025-01-28 by marq24

gorzan14, 2025-01-28:
In my opinion this bug is not away.

These values are only actual for first time integration ist initialized for some reason.
When a new car is attached the new plan, which I can see in the evcc frontend isn't updated to entities. See my previous post.
In Rest-API I can See the plan too: GET http://evcc:7070/api/loadpoints/1/plan        
`
{
    "result": {
        "planId": 2,
        "planTime": "2025-01-29T07:00:00+01:00",
        "duration": 4461,
        "plan": [
            {
                "start": "2025-01-28T23:00:00+01:00",
                "end": "2025-01-29T00:00:00+01:00",
                "price": 0.30440199999999995
            },
            {
                "start": "2025-01-29T03:00:00+01:00",
                "end": "2025-01-29T03:14:21+01:00",
                "price": 0.31773
            }
`
At you integration in home assistant, even after reload of the Integration it doesn't update

[image]

What do you think ist the Problem?

marq24, 2025-01-28:
The integration is using only GET 'http://evcc:7070/api/states' endpoint and there the info (SOC based) for vehicle plan is used (when known VEH is connected to LP) - and if it's a guest vehicle, then the PLAN data (energy based) from the loadpoint object is used

gorzan14, 2025-01-28:
OK. Now it gets more clear. I tried this too: GET 'http://evcc:7070/api/states'

When I use the one static Plan. Your Integration updates and shows values. Rest shows this:
               "effectivePlanId": 1,
                "effectivePlanSoc": 100,
                "effectivePlanTime": "2025-01-29T05:00:00Z",

When I use recurring plans, what we use usually your Integration does not update. But Rest shows the new plan in the same Variables!!
Just the plan Id changed.
                "effectivePlanId": 2,
                "effectivePlanSoc": 90,
                "effectivePlanTime": "2025-01-29T07:00:00+01:00",

As far as I checked it shows only the next/active/effective plan. So you could write it to the HASS entity regardless of planID.

What I don't understand are the different timeformts used by cvcc. PlanID is 6:00 in my time zone, Plan 2 is 7:00 in my timezone.

gorzan14, 2025-01-28:
This is more a Feature Request:

I would love if you could make this to values from GET 'http://evcc:7070/api/states' available as entities:

                "planProjectedEnd": "2025-01-29T03:54:13+01:00",
                "planProjectedStart": "2025-01-28T23:00:00+01:00",

Actualy I'm trying to get them with RESTfull-Sensors. But I am not that experienced. Could take some hours. Lets see who is faster ;-)