marq24/ha-evcc #96: sensor.evcc_carport_charge_total_import does not hold imported energy, but totalt energy usage

timpetersson81, 2025-02-19:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [x] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repro will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

sensor.evcc_carport_charge_total_import does not hold imported energy, but totalt energy usage for the charge.

It would be awesome to have the grid import, and perhaps the usage of solar energy in kWh. It seems that the sensor.evcc_pv_energy hold the totalt amount of solar energy for this day.

[image]
[image]

### Add your DEBUG log output

```shell
No loginfo, as the error is that data in the sensor is not as "promised"
```

marq24, 2025-02-19:
this data comes from the LOADPOINTS sesction...

## API:state:chargedEnergy -> Charge power
total changed energy ?! in WATT

## API:state:chargeTotalImport -> Grid imported 
total imported energy ?! - if this is not the case --- then provide an alternative...

timpetersson81, 2025-02-19:
It was not to say you work was bad, it was just the fact, I thought that something was messed up in the code somewhere, perhaps added wrong values to the sensor.

marq24, 2025-02-19:
this is not the case - you can check the output of the api/state endpoint what data your system is providing... I understand, that the lable's might not match (I picked the titles based on the value names in the API... ) when `chargeTotalImport` is actually not really the amount of energy that was imported from the grid, then indeed the name is not smart... but then the naming in the API is not smart...

timpetersson81, 2025-02-19:
Du you know if there is a Way to get amout charged in solarenergy?

UDicke, 2025-02-19:
There are some sensors provided by evcc and consequently by this app regarding the loadpoints. 
[image]
I think that the chargeTotalImport from the loadpoints provide the total energy transported through the loadpoint regardless of the source. Or "grid" in this case is the site...
See this, I installed this loadpoint in July 2024

[image]

There is another sensor called (sence: amount of pv usage). I guess this is the value for the last session and not for the whole life of the loadpoint. But all of this is "evcc" and not this integration...

[image]

There is another value provided by evcc, but this the site and not a loadpoint. I didn't check weather this is transported by this integration because I am not interested in this...

[image]

closed 2025-02-20 by marq24