marq24/ha-evcc #98: neues Objekt "Solar Prognose" hat die meiste Zeit den Wert "unbekannt"

walburgf, 2025-02-21:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [ ] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repro will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

Das neue Objekt "Solar Prognose" hat die ersten 5 min in jeder Stunde einen Wert, der Rest der Zeit ist der Wert des Objektes "unbekannt".
Der Wert des Objektes sollte bis zur Änderung den Wert behalten.

[image]

### Add your DEBUG log output

```shell

```

marq24, 2025-02-21:
was ist denn die response vom `http://[EVCC-IP]/tariff/solar` ?

walburgf, 2025-02-21:
die URL ist http://[EVCC-IP]:7070/api/tariff/solar (mit /api)
Dort erhalte ich die korrekten Werte.....

```
{
  "result": {
    "rates": [
      {
        "start": "2025-02-21T00:00:00+01:00",
        "end": "2025-02-21T01:00:00+01:00",
        "price": 0
      },
      {
        "start": "2025-02-21T01:00:00+01:00",
        "end": "2025-02-21T02:00:00+01:00",
        "price": 0
      },
      {
        "start": "2025-02-21T02:00:00+01:00",
        "end": "2025-02-21T03:00:00+01:00",
        "price": 0
      },
      {
        "start": "2025-02-21T03:00:00+01:00",
        "end": "2025-02-21T04:00:00+01:00",
        "price": 0
      },
      {
        "start": "2025-02-21T04:00:00+01:00",
        "end": "2025-02-21T05:00:00+01:00",
        "price": 0
      },
      {
        "start": "2025-02-21T05:00:00+01:00",
        "end": "2025-02-21T06:00:00+01:00",
        "price": 0
      },
      {
        "start": "2025-02-21T06:00:00+01:00",
        "end": "2025-02-21T07:00:00+01:00",
        "price": 0
      },
      {
        "start": "2025-02-21T07:00:00+01:00",
        "end": "2025-02-21T08:00:00+01:00",
        "price": 46
      },
      {
        "start": "2025-02-21T08:00:00+01:00",
        "end": "2025-02-21T09:00:00+01:00",
        "price": 416
      },
      {
        "start": "2025-02-21T09:00:00+01:00",
        "end": "2025-02-21T10:00:00+01:00",
        "price": 881
      },
      {
        "start": "2025-02-21T10:00:00+01:00",
        "end": "2025-02-21T11:00:00+01:00",
        "price": 1256
      },
      {
        "start": "2025-02-21T11:00:00+01:00",
        "end": "2025-02-21T12:00:00+01:00",
        "price": 1517
      },
      {
        "start": "2025-02-21T12:00:00+01:00",
        "end": "2025-02-21T13:00:00+01:00",
[... 161 lines cut ...]
      },
      {
        "start": "2025-02-22T21:00:00+01:00",
        "end": "2025-02-22T22:00:00+01:00",
        "price": 0
      },
      {
        "start": "2025-02-22T22:00:00+01:00",
        "end": "2025-02-22T23:00:00+01:00",
        "price": 0
      },
      {
        "start": "2025-02-22T23:00:00+01:00",
        "end": "2025-02-23T00:00:00+01:00",
        "price": 0
      },
      {
        "start": "2025-02-23T00:00:00+01:00",
        "end": "2025-02-23T01:00:00+01:00",
        "price": 0
      }
    ]
  }
}
```

marq24, 2025-02-21:
Funny! - that's all what I can say right now - after a 9h day @work I'll need some rest - but I can confirm that I am f*** stupid :-D

marq24, 2025-02-21:
please update to 2025.2.4 & thanks for your report!

closed 2025-02-21 by marq24