marq24/ha-evcc #100: Use Vehicle Data from EVCC for HomeAssistant

markaragnos81, 2025-02-25:
### Description

Hi. I use EVCC and Homeassistant with 2 vehicles. I have configured each vehicle in both EVCC and HomeAssistant. As a result, two different things are querying the APIs, and there's a risk of exceeding the query quota. Is it possible to use the vehicles configured in EVCC and the resulting values in HomeAssistant?

### API Keys

```shell

```

marq24, 2025-02-25:
Watt?? - I am sorry - but I did not pick up at all what is your current challenge you would liked to get solved... Probably you should describe more in detail your current setup - and most important on which side (EVCC-Server or HA-Server) you expect to reach a query quota?

markaragnos81, 2025-02-25:
Ok. Gerne auch auf Deutsch :-)
Ich denke, man könnte auch die Daten der in EVCC konfigurierten Fahrzeuge im HA nutzen um nicht von 2 Seiten immer die API's bezüglich Akkustand usw. anzufragen. Aktuell kann ich ja "nur" die Werte der Loadpoints und allgemeine Dinge im HA einsehen und konfigurieren.

marq24, 2025-02-25:
I have to admit, that I do not see any "current vehicle status" data in my sample API state endpoint data (which is already available from EVCC in HA) - this three:
 
- VEHICLEODOMETER (Fahrzeug Kilometerstand)
- VEHICLERANGE (Fahrzeug Reichweite)
- VEHICLESOC (Fahrzeug Ladestand)

This data comes from the current connected vehicle (to the loadpoint)... So if you have/see more vehicle related data - please be so kind and let me know, which EVCC API endpoint / state object this data can be found.

UDicke, 2025-02-28:
> Ok. Gerne auch auf Deutsch :-) Ich denke, man könnte auch die Daten der in EVCC konfigurierten Fahrzeuge im HA nutzen um nicht von 2 Seiten immer die API's bezüglich Akkustand usw. anzufragen. Aktuell kann ich ja "nur" die Werte der Loadpoints und allgemeine Dinge im HA einsehen und konfigurieren.

Hallo @markaragnos81 die Autos in HA und evcc abzufragen hat bei mir (VW und Mercedes) noch nicht zu Problemen geführt. In den entsprechenden HA Integrationen hast Du viel mehr Infos (brauch man nicht wirklich alle) und die Infos sind dauerhaft vorhanden, unabhängig davon ob das Fahrzeug verbunden ist oder nicht. In evcc sieht man den aktuellen Ladestand z.B. ja nur wenn das Auto mit einem Loadpoint per Kabel verbunden ist.
Ich glaube man kann evcc auch irgendwo sagen es möge die Autos abfragen auch wenn die nicht eingesteckt sind, aber warum das, selbst wenn es gehen sollte? Ich bleibe in jedem Fall bei "doppelter Konfiguration und Abfrage" solange es keinen Grund gibt das zu ändern. Und den sehe ich aktuell nicht...

ID3 in HA - gerade nicht eingesteckt...
[image]

Autos in evcc

[image]

markaragnos81, 2025-02-28:
Hallo @UDicke,

danke für deine Info. Mein Hintergrund sind die dauerhaft anhaltenden Probleme mit meinem Cupra Leon. Trotz 5m Cache hatte ich tatsächlich ab und zu Quota Probleme. Das hat sich jetzt aber grade komplett erledigt, da die Seat Integration nicht mehr gewartet wird und keiner die API Umstellungen für den Login machen kann :-(

UDicke, 2025-02-28:
Welche Seat Integration meinst Du, HA oder evcc?
Oder die Seat Hersteller api?

markaragnos81, 2025-02-28:
Die in HA. https://github.com/farfar/homeassistant-seatconnect
Da geht leider nichts mehr. Kennst du ggf. eine Alternative ?

UDicke, 2025-02-28:
Nein, aber da gibt es bestimmt irgendwo eine Lösung. Du bist ja nicht der einzige betroffene Anwender. 
Schau doch mal hier...
https://github.com/Farfar/homeassistant-seatconnect/issues/80#issuecomment-2679633965

marq24, 2025-03-14:
ich mach' hier das issue ma' zu...

closed 2025-03-14 by marq24

UDicke, 2025-03-14:
@markaragnos81 läuft denn evcc bei Dir nach wie vor? Da ist der Skoda ja im Gespräch...
https://github.com/evcc-io/evcc/pull/19523
Dann sollte das Auto auch in der Integration zu sehen sein wenn es eingesteckt ist.