marq24/ha-evcc #102: Implement new "Reset Cache" API request

Schnabulation, 2025-02-27:
### Description

In the latest release of evcc a new API request has been implemented:

https://github.com/evcc-io/evcc/pull/18941

Would be great if you could implement this request for easy access.

### API Keys

```shell

```

Schnabulation, 2025-02-27:
To add a little more context: I requested this new feature over at evcc so we could achieve the following:

In Home Assistant add a script that does the following:

- Start to preheat the car via own car integration
- Reset the car cache in evcc
- Have evcc re-poll the car and find out that it's preheating
--> evcc starts charging to preheat from wallbox

marq24, 2025-02-27:
more details are required - AND if this new API endpoint will require additional auth at EVCC, then I don't see this coming -> then I suggest a REST template for this "button-press-action"...

Schnabulation, 2025-02-27:
The request is kinda new, no documentation available at the moment, I'm sorry. What more details do you need?

Does your integration not support API requests with AUTH at the moment? If so I'll try to implement this as a REST command.

marq24, 2025-02-27:
yes - currently NO-AUTH is required to access the evcc entpoints - thus no auth flow is currently implemented (there are no credentials you must provide during the initial setup of the integration)...

closed 2025-02-27 by Schnabulation

SolarPower2024, 2025-03-01:
@Schnabulation as someone who is not so familiar with Rest commands: did you manage it to set up and could you maybe provide your HA code?