marq24/ha-evcc #105: Error adding the URL

aravindnadumane, 2025-03-09:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [x] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repro will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

Hello Team,
 during the adding the integration in the Setting/ Devices & Integration, I get the Error "URL is wrong - could not reach evcc-system" and cannot proceed with the Setup. I tried the Latest Version 2025.3.1 as well as going back to an older version 2025.2.3.

Any help what it could be ? I use the local IP of my Home assistance, 

[image]

### Add your DEBUG log output

```shell

```

marq24, 2025-03-09:
looks like you have an issue with 'reading' 

> Das Schwarze ist die Schrift...

closed 2025-03-09 by marq24

aravindnadumane, 2025-03-09:
Thanks for the gentle Answer. Although it didn't help me either.

Of course i ticked the first block else i couldn't even post my problem.
I'm not an expert in  HA and merely configured mentioned in this tutorial: https://www.youtube.com/watch?v=8v2Z8ynACPE&t=365s

marq24, 2025-03-09:
[image]