marq24/ha-evcc #107: Value of individual solar sources Unknown, total still known after upgrade to 2025.3.3

mpw96, 2025-03-16:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [ ] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repro will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

After upgrading this integration from 2025.3.2 to 2025.3.3 the sensors sensor.evcc_pv_0_power and sensor.evcc_pv_1_power have an unknown value. But sensor.evcc_pv_power still has a value. When I go back to 2025.3.2 the individual ones have values again.

[home-assistant_evcc_intg.log](https://github.com/user-attachments/files/19269489/home-assistant_evcc_intg.log)

### Add your DEBUG log output

```shell

```

mpw96, 2025-03-16:
2025.3.2:
[image]

2025.3.3:
[image]

mpw96, 2025-03-16:
My (automated) upgrade proecedure is to remove the custom_components/evcc_intg folder, put the new version in place and restart Home Assistant.

mpw96, 2025-03-16:
EVCC Version is 0.200.9, even though the debug log reports 0.131.1. This might have been the version at the time when I started to use your integration (which is awesome!).

mpw96, 2025-03-16:
EVCC delivers it:
```
curl http://192.168.96.120:7070/api/state|jq
...
    "pv": [
      {
        "power": 1299,
        "energy": 3038.44
      },
      {
        "power": 101.7,
        "energy": 1906.409
      }
    ],
    "pvEnergy": 4944.849,
    "pvPower": 1400.7,
...
```

marq24, 2025-03-16:
... are you using websocket? If yes, can you disable it and restart HA? 

And if this is the root cause - can you capture (via a regular browser dev tools) the message containing the pv data?

mpw96, 2025-03-16:
After going back from 2025.3.3 to 2025.3.2 (back to the working setup) I have two devices:
[image]

When upgrading to 2025.3.3 I have only one device. I don't know if this information helps or confuses...

marq24, 2025-03-16:
Please be so kind and update to https://github.com/marq24/ha-evcc/releases/tag/2025.3.4 - not the data container can contain also arrays - but the code (introduced with 2025.3.3 always checked for dicts - at least that's most plausible error cause for me...

marq24, 2025-03-16:
The duplicated device issue - is expected (when you switch back from 2025.3.3 to any older release: In order to support multiple instance of the integration I had to modify the device-identifier - thus 2025.3.3 will create a new device... But it also includes 'evcc-legacy device cleanup code' - with every start of 2025.3.3 (or higher) old evcc devices will be removed from the system - you can see the difference by comparing the Manufacturer string: new is just 'marq24'....

mpw96, 2025-03-16:
It works after upgrade to 2025.3.4. Thank you for the lightning-fast fix! :hourglass_flowing_sand:
It feels like I could watch you coding!

closed 2025-03-16 by mpw96