marq24/ha-evcc #112: All entities are Unknown

mortzel, 2025-03-21:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [ ] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repo will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

Added the evcc connection to a running evcc without a vehicle or wallbox but with pv and all entities in home assistant are in the state unkown. When i stop the evcc container the entities are unavailable so the connection is working.

### Add your DEBUG log output

```shell
Error doing job: Task exception was never retrieved (None)

Traceback (most recent call last):
  File "/config/custom_components/evcc_intg/sensor.py", line 153, in native_value
    value = self.coordinator.read_tag(self.tag, self.idx)
  File "/config/custom_components/evcc_intg/__init__.py", line 428, in read_tag
    ret = self.read_tag_statistics(tag=tag)
  File "/config/custom_components/evcc_intg/__init__.py", line 441, in read_tag_statistics
    if tag.subtype in self.data[JSONKEY_STATISTICS]:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 268, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 479, in _async_refresh
    self.async_update_listeners()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 178, in async_update_listeners
    update_callback()
    ~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1023, in async_write_ha_state
    self._async_write_ha_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1148, in _async_write_ha_state
    self.__async_calculate_state()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1085, in __async_calculate_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1029, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 520, in state
    value = self.native_value
            ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/evcc_intg/sensor.py", line 206, in native_value
    _LOGGER.debug(f"tag: {self.tag} (idx: '{self.idx}') (value: '{value}') caused {err}")
                                                                  ^^^^^
UnboundLocalError: cannot access local variable 'value' where it is not associated with a value
```

marq24, 2025-03-22:
thanks - please update to https://github.com/marq24/ha-evcc/releases/tag/2025.3.5 and close this issue when it's fixed

mortzel, 2025-03-22:
Yeah. I have configured a grid, pv and battery meter also 2 tarrifs, but it looks like you are correct. The webinterface shows data for all three meters but there is no data in the output of `/api/status`

[image]
[image]

closed 2025-03-22 by mortzel

marq24, 2025-03-22:
In any case the integration should be able to deal with data object not set (yet) - actually I ran into the same error during the week - after a complete home-server reboot - and "fixed" it already (but for what ever reason (aka "lame excurse for getting seriously old") I forgot to create a release ;-)