marq24/ha-evcc #113: number.evcc_battery_grid_charge_limit wird in HA nicht aktualisiert

jove01, 2025-03-22:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [ ] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [ ] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repo will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add your DEBUG log output

```shell

```

jove01, 2025-03-22:
In EVCC Hausbatterie-Ladelimit geändert

[image]

jedoch in HA keine Aktualisierung

[image]

Benötigst du zusätzliche Info?

marq24, 2025-03-22:
Entweder aktiviert Du die Websockt Kommunikation - oder Du musst in HA mal bis zu 5min warten meine ich, bis der Wert mal wieder vom Backend geholt wird... (für diverse Config Werte ist der sync langsamer)... 

Da bei mir der Wert 'batteryGridChargeLimit' gar nicht in EVCC mehr auftaucht, musst ich erstmal eine neue Batterie konfigurieren - danach musst ich ich je einmal EVCC und HA neu starten - dann war der Wert aber auch bei mit HA da... [sowohl mit als auch ohne WebSocket]...

jove01, 2025-03-22:
Websocket war aktiviert. Nach der Deaktivierung war der Wert in HA zu sehen. Eine Änderung benötigte sehr lange im HA )obwohl im MQTT-Explorer sofort zu sehen war).
Mit der erneuten Aktivierung des Websockets war dann auch eine Änderung in HA schnell zu sehen.

Für mich wirkt das fast so, als ob ein initialer Austausch auf Entitätenebene benötigt wird. Sollte es auch bei anderen Entitäten Probleme weiß ich, dass ich die Einstellung mal ändern sollte.
Vielen Dank

closed 2025-03-22 by jove01