marq24/ha-evcc #116: Change evcc URL after integration has been set up

jeriox, 2025-03-27:
### Description

After the integration worked flawlessly for me for a long time, it suddenly stopped working, complaining that ` UpdateFailed unexpected: Cannot connect to host power.local:7070 ssl:False [DNS server returned answer with no data] `. I don't understand why this is happening, because all other hosts in the network happily resolve the name and the homeassistant host also did so until recently. However, I just wanted to avoid the issue by changig the host to the proper IP address. However, when clicking "Configure" for the integration, there is no option to change the URL. I would appreciate that option to avoid deleting and recreating the integration and fixing everything that is related to it.

### API Keys

```shell

```

marq24, 2025-03-27:
Hi Julian,

host change is not supported via config - the simples way is to remove your current integration and just re-add it using the IP...

marq24, 2025-03-27:
please update to https://github.com/marq24/ha-evcc/releases/tag/2025.3.7 and close the issue, when the enhancement is implemented

closed 2025-03-31 by jeriox