marq24/ha-evcc #118: Temparature for Heating

hbpv, 2025-04-09:
### Description

Hi

I'm using your integration for a wallbox AND heating device.

Would it be possible to show ° Celcius for entities instead of % for SOC/Max SOC when using heating products with evcc like MY-PV? In mqtt evvc is publishing that the certain device is a "heating" device.

Should apply to:

effective_limit_soc
vehicle_soc
effective_plan_soc

[image]

Best,
Max

### API Keys

```shell

```

marq24, 2025-04-09:
let me think about, IF and how this is possible to support... I have no "quick" answer for you right now...

marq24, 2025-04-11:
imho there are more ?!

VEHICLE_MIN_SOC
VEHICLE_LIMIT_SOC
VEHICLE_PLANS_SOC

hbpv, 2025-04-11:
> imho there are more ?!
> 
> VEHICLE_MIN_SOC VEHICLE_LIMIT_SOC VEHICLE_PLANS_SOC

Ah yes, correct!

marq24, 2025-04-11:
update to https://github.com/marq24/ha-evcc/releases/tag/2025.4.1 and close if sloved

hbpv, 2025-04-11:
Hey. lot of entities are no longer avaible. Only error i found:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/config/custom_components/evcc_intg/sensor.py", line 51, in async_setup_entry
    a_stub.Tag == Tag.EFFECTIVEPLANSOC or
    ^^^^^^^^^^
AttributeError: 'ExtSensorEntityDescriptionStub' object has no attribute 'Tag'. Did you mean: 'tag'?

marq24, 2025-04-11:
upsi - sorry - I created a patched version... (you might to select reinstall manually in HACS)

hbpv, 2025-04-11:
> upsi - sorry - I created a patched version... you might to select reinstall manually ("Erneut herunterladen") in HACS

Kein Thema, aber die Werte sind weiterhin %

marq24, 2025-04-11:
... ich seh's auch gerade lokal im SIMULATOR :-/

marq24, 2025-04-11:
patch II - alle guten Dinge sind 3 ?

hbpv, 2025-04-11:
Sieht gut aus, danke dir, wirklich super die schnelle Lösung!

closed 2025-04-11 by hbpv

guineapirates, 2025-04-12:
Hi

Danke für die Anpassung!
Habe aber das Problem dass "limit_soc" immer noch in % angezeigt wird.
Der Rest hat sich korrekt auf "°" geändert....

marq24, 2025-04-12:
uuuund noch ein Grund - limit_soc ist ja gar kein Sensor - sondern eine Number - da muss ich dann auch noch ein bissel dran rum schrauben...