marq24/ha-evcc #125: sensor.evcc_<loadpoint>_charge_current has unknown value since evcc 0.203.1

mpw96, 2025-04-20:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [ ] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repo will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

After upgrading to evcc 0.203.1 the above sensor has value "unknown". I restarted both evcc and this integration - it did not help.

### Add your DEBUG log output

```shell
marc-philip@birne:~$ curl -X GET http://evcc-server:7070/api/state|jq|grep -i current
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 19134    0 19134    0     0  1281k      0 --:--:-- --:--:-- --:--:-- 1334k
      "currents": [
        "chargeCurrents": [
        "effectiveMaxCurrent": 16,
        "effectiveMinCurrent": 6,
        "maxCurrent": 16,
        "minCurrent": 6,
        "offeredCurrent": 16,

Which entry are you reading? Is it still there?
```

mpw96, 2025-04-20:
Okay, shame on me. I did not have version 2025.4.1 ha-evcc and you are using offeredCurrent. Ich wünsche dicke Eier!

closed 2025-04-20 by mpw96

mpw96, 2025-04-20:
https://github.com/evcc-io/evcc/pull/20457

marq24, 2025-04-20:
was soll ich sagen :-) - manchmal komme ich mit den ganzen Änderungen in EVCC nicht wirklich hinterher ... aber bei diesem Change hatte ich mal wieder die Nase vorn... Nur leider ist es wohl so, dass HASS nicht immer eben rechtzeitig über Updates informiert...