marq24/ha-evcc #127: Read all cost data / history from EVCC

loryanstrant, 2025-04-23:
### Description

It would be nice instead of just the charging session cost, we could actually see additional cost statistics like total for the day, week, month, year, etc.
Given that EVCC has a list of sessions and their costs, could this be used?

### API Keys

```shell
api/sessions
```

loryanstrant, 2025-04-23:
I built a crude custom component to test the theory. It works, but the sensors could be improved to give them IDs and other aspects.

[evcc_historical_data.zip](https://github.com/user-attachments/files/19859417/evcc_historical_data.zip)

marq24, 2025-04-23:
So yes - technically it's possible to add api/sessions endpoint requests to the current implementation... you sample code looks reasonable to me... so why not trying to integrate it here (via a PR?)

loryanstrant, 2025-04-23:
Because I built it using ChatGPT and don't actually know how to code in Python, so I'm not game to try to modify what you've built to insert my code. :-)

loryanstrant, 2025-04-24:
Heh, all good, I understand.
I was more doing it to validate that the data could be obtained.
I'm not a developer, but I do work in other languages. Python is just not something I have the brain space to take onboard. :-)

marq24, 2025-04-24:
> I was more doing it to validate that the data could be obtained.

that's highly appreciated!

> Python is just not something I have the brain space to take onboard. :-)

It's just code like any other - c++, java, node, c#  - at a certain level they are all the same... well same same but different... I have started with python cause of a HA integration that I wanted to improve...

loryanstrant, 2025-04-26:
I hear you, I just where I can go down a rabbit hole and burn too much time (which I already do with plenty of other things, thanks to ADHD). 😄

superpower10, 2025-07-29:
Any news when it will come?

marq24, 2025-09-06:
"today" is the day... But you need to create your own template sensors to access vehicle or loadpoint data

loryanstrant, 2025-09-06:
That's awesome, thanks so much!
Does the new sensor only capture charging session data from the point it was created, or historical?
Because the only attribute I can see in the charging sessions sensor is the total:

[image]

marq24, 2025-09-07:
@loryanstrant can you please share your JSON response of the `/api/sessions` request with me (eMail) ?! Looks like that something in the data crashing the "create-summ" code - there might be already a hint in the HA log ?!

TIA

marq24, 2025-09-07:
before that - just update to 2025.9.1 and let's see if this will fix the issue already...

loryanstrant, 2025-09-07:
I've updated to 2025.9.1 and restarted HA, still the same issue with the charging sessions sensor only showing number of sessions.
I've got the JSON response from the /api/sessions data, am happy to send it to you - what email can I use?
Interestingly not all of the charge sessions have "evcc" in the identifier, so I wonder if that could be causing it?

marq24, 2025-09-07:
you find my email on my github profile

marq24, 2025-09-08:
@loryanstrant - thanks for the data - I already released an update yesterday evening [2025.9.2] that finally fix your issue - there are two entries in your sessions, that contain "none-plausible" data:

`{'id': 325, 'created': '2025-08-16T08:59:35.048733861+10:00' ...` has no/empty `price`
and
`{'id': 11, 'created': '2025-03-08T10:27:55.986405118+01:00' ...` has no/empty `vehicle`

both will now be handled by the integration (2025.9.2) so you have the data now - you also might like to fix the entries via the evcc GUI (not 100% sure if the empty price can be fixed there)