marq24/ha-evcc #131: Not all power values updating

clonefx, 2025-05-11:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [ ] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repo will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

Hello,

first of all thanks for the integration!
I have the issue that now all power values are being updated, i.e. Only the home power sensor is updated according to the configured interval. Grid power is unknown and pv power values are updated only on (re-) start of the integration.
Evcc is showing and updating all values.
Any ideas or did I miss a certain configuration perhaps..?

Thanks! 

### Add your DEBUG log output

```shell

```

marq24, 2025-05-11:
are you using websocket feature: yes/no?

If no: is your car connected to the charger yes/no?

If no: by default only few values will be updated

clonefx, 2025-05-11:
Websocket: no
Car: connected and charging

So I guess I need to work with the available sensors. Thanks!!

marq24, 2025-05-11:
I just can recommend to enable the Websocket feature of the Integration...

marq24, 2025-05-11:
and just as additional info... (when websocket is disabled) then at least every 5minutes all data should be fetched by the integration from the evcc server (no matter if the update interval is smaller)...

Everytime you sould see in the (debug) log an entry like this: `going to read all data from evcc@...`

If you don't have this entries in your log -> please let me know!

marq24, 2025-05-12:
please update to latest version... 2025.5.0

clonefx, 2025-05-12:
Works perfectly after updating! Thanks a lot!

closed 2025-05-12 by clonefx