marq24/ha-evcc #132: battery SOC sensor is missing device class

krispek, 2025-05-11:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [x] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repo will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

While adding the SOC to a dasboard, I noticed that the device class of the `sensor.evcc_battery_soc` is missing. Consequently, HA does not show a nice battery icon that indicates the state of charge. Would it be possible to add `device_class: battery` to this sensor? The bug report states the current state attributes. 

### Add your DEBUG log output

```shell
state_class: measurement
unit_of_measurement: "%"
icon: mdi:home-battery-outline
friendly_name: Battery SOC
```

marq24, 2025-05-11:
at least you provided a possible solution...

[image]

... I am not 100% confident yet, if the soc of an external battery would be "the soc indicator for the evcc integration"... IMHO if you have a single LP instance, then I would consider the SOC of the connected vehicle more appropriate... 

But at least you can alter your local installation to fit you needs by declaring the `device_class` in the corresponding (Ext)EntityDescription ?!

krispek, 2025-05-11:
🙈 My bad. After the fifth bullet I stopped reading. 

I fixed it locally by adding the device class _and removing the icon_ in the attributes. Now it works fine on my instance, so don't hurry. 

I use this value when the car is connected to the charger. I allow the Bluelink integration to force update the values only every 24h to prevent the 12v battery form discharging. When the car is not charging (and most of the times that means driving), I use the Bluelink values, cause they are updated much more frequent and the EVCC values aren't. So that's best of both worlds. :)

marq24, 2025-05-11:
> I use this value when the car is connected to the charger. I allow the Bluelink integration to force update the values only every 24h to prevent the 12v battery form discharging. When the car is not [...]

?! I have to admit, I don't get the connection between _using the Home-Battery-SOC as central Battery-Info for the Integration (which IMHO adding the 'device_class' will cause)_ ...  and any values the `Hyundai Bluelink Integration` provide - or not provide ?!

I just can recommend to request vehicle data only from one place (either EVCC or HA - but not both) - and make the data available to the other system... either from evcc to HA or (and that's my pfeffernd way) from __HA to evcc__.

closed 2025-05-29 by marq24