marq24/ha-evcc #140: vehicle limit - SOC

balgerion, 2025-06-06:
### Description

Maybe I'm blind, but is there a vehicle limit in the integration? If not, is it possible to add one? :)

[image]

### API Keys

```shell

```

marq24, 2025-06-06:
one is for loadpoint and the other one is for the car...

marq24, 2025-06-06:
I'll check when I am back @ home what's causing here the confusion... might be that in the meantime something has changed internally in the evcc api response...

Even if it's quite a while ago - there is a soc_limit for the current charging process at the loadpoint and there is a SOC limit at vehicle level - and there is also `effective_limit_soc`... 

I must check, which sensor will extract which value from which part of the state-respone - but there is/was a loadpoint related and a vehicle related value... [as I said, I must check this in the code, which sensor extracts "what"]

marq24, 2025-06-07:
also ich hoffe wir sind uns schonmal über die Funktion von `number.evcc_[LP-ID-HERE]_limit_soc` einig?!

[image]

wie Ihr aber auch sehen könnt, habe ich bei "mir" erstmal keinen zweiten Wert neben dem 'Cent' Symbol - ich gehe aktuell davon aus, das Ihr eine Ladeplanung am start habt?

UDicke, 2025-06-07:
> wie Ihr aber auch sehen könnt, habe ich bei "mir" erstmal keinen zweiten Wert neben dem 'Cent' Symbol - woher kommt dieser Wert - einen "PLAN" ist nicht aktiv/am start... ist das die aktuelle evcc version (0.204.1) ?

Der Wert erscheint nur wenn das Limit unter 100 % ist. Es ist ein Limit das im Auto eingestellt wird und in evcc nur angezeigt wird.
bei mir z.B. heute eingestellt auf 90% zum Vorzeigen: 

[image]

[image]

ein Plan ist nicht involviert.

UDicke, 2025-06-07:
Als  effective_limit_soc wird das in evcc eingestellte Limit angezeigt, evtl. auch eins aus einem Plan, das hab ich nicht geprüft!

[image]

```
2 |  
batteryBoost | false
chargeDuration | 10215
chargePower | 2665
chargeRemainingDuration | 2676
chargeRemainingEnergy | 1982.5
chargeTotalImport | 2737.65
chargedEnergy | 6050
chargerFeatureHeating | false
chargerFeatureIntegratedDevice | false
chargerIcon | null
chargerPhases1p3p | true
chargerSinglePhase | false
chargerStatusReason | "unknown"
charging | true
connected | true
connectedDuration | 0
disableDelay | 180
disableThreshold | 0

effectiveLimitSoc | 95

effectiveMaxCurrent | 16
effectiveMinCurrent | 6
effectivePlanId | 0
effectivePlanSoc | 0
effectivePlanTime | null
effectivePriority | 2
enableDelay | 60
enableThreshold | 0
enabled | true
limitEnergy | 0

limitSoc | 95

maxCurrent | 16
.
.
.
smartCostNextStart | null
title | "C-Fos"
vehicleClimaterActive | false
vehicleDetectionActive | false

vehicleLimitSoc | 90

vehicleName | "GLC300E"
vehicleOdometer | 12494
vehicleRange | 94
vehicleSoc | 83.336
vehicleWelcomeActive | false
```

Und ich bin einer von denen der die Fahrzeuge in HA und evcc abfragt 🤔, ausser bei dem Ford Kuga unserer Tochter...😊
Vielen Dank übrigens für Deine Arbeit zu diesem Thema! 👍

marq24, 2025-06-07:
@balgerion - sorry for switching to German - I hope auto-translate will do the trick for you...

tkoeberl, 2025-07-29:
@marq24 : just wondered about the same like @snowdd1.
In my opinion the actual values can be read out by: 
api/state (vehicles.{name}.limitSoc AND vehicles.{name}.minSoc) 
and set by: /vehicles/{name}/limitsoc/{soc}  /vehicles/{name}/minsoc/{soc}
So its not connected to the cars vendor software, it is  evcc only.

marq24, 2025-07-29:
... IMHO the vehicle limit should be set via the API of the Vehicle - in my case this is possible via the FordPass integration... Yes I see the point that the current vehicle-soc-limit ist not a sensor in ha

pencoe, 2025-08-27:
sorry, have to switch to german for this...
Also, ich kann für ein Fahrzeug im Planer unter "Ankunft" sowohl ein Minimum SoC als auch ein Standard-ZielSoC einstellen. Sobal ein Fahrzeug angesteckt wird, wird dieser Wert von evcc auch gesetzt und im Status als effective_limit zurückgeliefert. Dieser Wert ist aber read only: deshalb müsste der einstellbare limit_soc auf diesen Wert gesetzt werden, sobald sich effective_limit ändert.
Ich bin jetzt nicht sicher, was passiert, wenn man in evcc den Wunsch-ZielSoC ändert, aber sobald ich manuell an dieser Einstellung in evcc "drehe", ändert sich auch der limit_soc-Schieber dieser Integration, nur halt beim Anstecken des Fahrzeugs passiert das nicht.

marq24, 2025-08-27:
also bei Dir ist es so, dass sich in EVCC der Wert von api/state effectiveLimitSoc auf 90% gesetzt wird, sich aber der Wert der Entität `number:evcc_[LP-NAME-HERE]_limit_soc` (Schieberegler) aus der evcc Integration __nicht__ nicht ändert ?!

snowdd1, 2025-08-28:
Ich habe mir eine Automatisierung geschrieben: Wenn Fahrzeug angesteckt wurde, setze number.evcc_lp1_limit_soc auf den sensor.evcc_lp1_effective_soc-Wert (den ich aus historischen Gründen via MQTT abgefragt hatte, ich glaube es gab den effective_soc damals nicht in dieser Integration. Funktioniert seit Juli diesen Jahres gut. Wäre aber natürlich schöner wenn der Wert in der Integration direkt gesetzt würde.

snowdd1, 2025-08-28:
Off topic: Nicht perfekt (insb nicht Pixelperfect) aber so sieht es bei mir aus:

[image]

snowdd1, 2025-08-28:
100 bei mir