marq24/ha-evcc #143: sensor.evcc_battery_mode  über API POST

jove01, 2025-06-15:
### Description

Ich habe den sensor.evcc_battery_mode als sensor aktiviert (sensor.evcc_battery_mode_value). Zeigt dann auch Veränderungen an.
Ich würde gerne den Mode "hold" zur Entladesperre setzen.
Kann dies über deine Integration umgesetzt werden?

### API Keys

```shell

```

marq24, 2025-06-15:
also ohne jetzt in den Code geguckt zu haben - `sensor.evcc_battery_mode` - also das ist keine number, switch oder select - damit ist das erstmal read-only... 

Worüber setzt Du denn die Einstellung 'hold' direct in evcc?

jove01, 2025-06-15:
Hier wird eine Möglichkeit beschrieben
[[https://github.com/evcc-io/evcc/wiki/aaa-Lifehacks#entladung-eines-steuerbaren-hausspeicher-preisgesteuert-oder-manuell-sperren](https://github.com/evcc-io/evcc/discussions/url)](url)

Über MQTT konnte ich den Mode nicht setzen.

Hier hatte ich wegen der Umsetzung nachgefragt
[https://github.com/evcc-io/evcc/discussions/21837](url)

Da ich trotz der dann erfolgreichen Umsetzung dass im EVCC-UI nicht optimal finde, würde ich das gerne selber per API (dort als Alternative genannt, mit meinen Kenntnissen aber problematisch) oder noch besser über die Möglichkeit deines Sensors machen.

jove01, 2025-06-15:
Das ist nur die Voraussetzung
[https://github.com/evcc-io/evcc/discussions/21837#discussioncomment-13470850](url)

https://github.com/evcc-io/evcc/discussions/21837#discussioncomment-13471397

Und hier der letzte Absatz!

Wie gesagt, ich hatte den Loadpoint eingerichtet und die Sperre mit "Schnell" aktiviert. Die Info "Gesperrt" war dann im UI zu sehen und im MQTT-Explorer der Mode "hold"

marq24, 2025-06-15:
... tut mir leid - ich kann Dir hier nicht folgen - ich kann was mit API-Endpoints und oder JSON Objekten aus dem Status Endpoint etwas anfangen... mir aus n-Threads irgendwie "blind" die Richtigen Informationen rauszusuchen übersteigt aktuell meine Kapazitäten...

marq24, 2025-06-15:
ja natürlich habe ich die REST API im Zugriff... da ich aber selbst kein steuerbaren Speicher mein Eigen nenne, fällt es mir schwer aus den hier zusammengetragenen Fragmenten eine für mich umsetzbare Aufgabe abzuleiten.

Also welchen aktuellen Schalter der in evcc existiert möchtest Du auch von HA aus schalten? - Und wenn es Dir bekannt ist, welcher API Endpoint wird dafür verwendet - UND gibt es Voraussetzungen, das ein solcher Schalter überhaupt in der evcc GUI angezeigt wird...

jove01, 2025-06-15:
Hier der Ladepunkt und Charger gemäß Hack

==> Ergebnis ganz unten

nur ist es eleganter und flexibler, den batterymodus per Automatisierung zu setzen!!

[image]

[image]

Status ohne Sperre

[image]

Und jetzt auf Schnell

[image]

Ergebnis

[image]

marq24, 2025-06-15:
mhhh - also ich will mich ja nicht dümmer anstellen als ich Dir sicherlich erscheine... aber aus der Sequenz dieser Screenshots werden ich leider nicht schlau... 

Das einzige was ich tatsächlich erkenne ist, das es im letzten Screenshot eine anzeige gibt, dass die Batterie 'gesperrt' ist - warum dies so ist - also welche Einstellung in aus den Screenshots zuvor jetzt aus dem "normalen" Batteriemodus den 'gesperrten' macht, verstehe ich einfach nicht... 

Also was passiert zwischen 'Status ohne Sperre' und dem 'Ergebnis' ?...  Also es soll also einen Schalter geben, ob im SCHNELL-Modus die Batterie mitverwendet werden kann oder nicht??!

Und eine Bestätigung, dass es ausschließlich um den SCHNELL Modus geht... wäre dann auch noch recht hilfreich...

marq24, 2025-06-15:
denn dann reden wir hier von 'diesem' Schalter...

[image]

jove01, 2025-06-15:
leider nein

Grund der Ladesperre ist:

a)
[image]
b)
beim sogenannten Netzladen, also Befüllen des Hausspeichers mit günstgem Netzstrom

ich versuche meine Fragen zu dem Lifehack im Zusammenhang mit der Entladesperre und die wichtigsten Antworten dazu zusammenzufassen; siehe
[https://github.com/evcc-io/evcc/discussions/21837](url)

Voraussetzung für die im Standard installierte Ladesperre ist dieser Schalter ganz unten

[image]

EVCC speicher die Sperre dan in
`Battery mode. 0: unknown, 1: normal, 2: hold, 3: charge`
mit dem Wert **hold**

Die Anzeige im UI hatte ich oben gezeigt.

Der sogenannte Lifehack dient als Krücke, die Sperre z.B. bei günstigem Strompreis über EVCC zu setzen.

Zusammengefasst wurde dort geantwortet

> Entweder mit dem speziellen Ladepunkt und dessen Umschaltung auf schnell, oder die Einstellung des smartCostLimit für eben den Ladepunkt - Vorteil: weniger externe Logik nötig. Alternativ kannst du das extern per HA oder anderer Automatisierung auch direkt per batteryMode machen.

oder eben per API

[image]

Ich hoffe, die Problematik verständlich dargestellt zu haben.

marq24, 2025-06-15:
> Ich hoffe, die Problematik verständlich dargestellt zu haben.

kurz und knapp - leider nein...

jove01, 2025-06-15:
ok. Ich würde mich dann nur wiederholen.
Für mich bedeutet das, dass ichmich  für eine Lösung entscheiden muss:
a) per MQTT diesem Loadpoint aus dem Hack den Befehl gebe, die Batterie zu laden
b) mich in Rest API einarbeite

Vielen Dank

SolarPower2024, 2025-07-12:
@jove01 das was du realisieren möchtest läuft über die Rest API
/batterymode/{batteryMode}

Da kannst du den Status "charge", "hold" oder "normal" setzen.
Laut Doku https://docs.evcc.io/docs/integrations/rest-api  setzt sich dieser aber alle 60 Sekunden zurück, heißt du musst den Befehl dann regelmäßig schicken.

jove01, 2025-07-12:
> [@jove01](https://github.com/jove01) das was du realisieren möchtest läuft über die Rest API /batterymode/{batteryMode}
>
> Da kannst du den Status "charge", "hold" oder "normal" setzen. Laut Doku [...]

Vielen Dank
Ich hatte Probleme mit der API, daher habe ich das so gelöst
`      - data:
          topic: evcc/site/batteryMode/set
          payload: hold
        action: mqtt.publish
      - delay: 00:00:30`

closed 2025-07-12 by jove01