marq24/ha-evcc #145: Connection to Docker evcc instance refused

starsign78, 2025-07-03:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [ ] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repo will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

Configuring the evcc instance doesn't work for me:

[image]

Yes, I can access the evcc instance via browser and mobile app:

[image]

Home Assistant and evcc are running in separate Docker containers on Synology.
- Home Assistant on host network
- evcc on macvlan network
-> could that be the issue (containers not being able to communicate to each other while both being reachable from other clients)?

### Add your DEBUG log output

```shell
2025-07-03 13:59:42.266 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] going to read all data from evcc@http://192.168.0.201:7070
2025-07-03 13:59:42.267 DEBUG (MainThread) [custom_components.evcc_intg.pyevcc_ha] GET request: http://192.168.0.201:7070/api/state
2025-07-03 13:59:45.274 ERROR (MainThread) [custom_components.evcc_intg] Exception while test credentials: Cannot connect to host 192.168.0.201:7070 ssl:False [Connect call failed ('192.168.0.201', 7070)]
```

marq24, 2025-07-03:
You might like to consider using the docker host option?

https://docs.docker.com/engine/network/tutorials/host/

or when you use docker-compose: `network_mode: host`

starsign78, 2025-07-03:
I solved it the other way round and moved the ha container into the macvlan network (like the evcc instance). This way both containers can access each other and have their distinct IPs.
But yes, having both on the host network would've probably worked too :-)