marq24/ha-evcc #158: After restart no connection first.

cracyfloyd, 2025-08-03:
### Checklist

- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [x] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repo will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

### Add a description

no connect 1st time

### Add your DEBUG log output

```shell

```

cracyfloyd, 2025-08-03:
After restarting Home Assistant, the instance does not connect to the EVCC server. Only a manual restart of the instance resolves the issue. WebSocket is enable.

bierma1966, 2025-08-03:
I ' ve got the same issue.
I take a full reboot of my RPI 5 Every night at 1:00 .
Home Assistent is full reboot but the EVCC Addon needs about 5min. more to start the server.
I solved by create another Automation to restart HA Core Integration.

marq24, 2025-08-03:
OF course evcc server must be running before you start the integration - in e.g. in docker-compose you can define dependecies between services .

closed 2025-08-03 by marq24

cracyfloyd, 2025-08-03:
I have now also solved it with automation. However, it would be better if the instance checked whether the EVCC is online and then connected. But it also works with automation. thnx all.

marq24, 2025-08-05:
> Cannot understand why this bug is closed and not corrected. Every user needs to fiddle around by a kind of workaround automation.

I can't understand so many things - specially the exceptions of human beings, seams to have little to none reading abilities... nor the required minimum knowledge how to operate infrastructure...

> ### Checklist
> [x]  I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA[x]  I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainers of this repo will spend time explaining the world to me - that's their job!. | I live by the motto: Better to ask than to think about it once. | It's the maintainers' own fault that they provide open-source software and are willing to offer free support.

This HomeAssistent Integration have the __requirement__ of an running evcc server instance to be able to start & work [mainly cause during startup the evcc configuration is required to configure the right amount of entities per loadpoint and other dynamic stuff].

That you reboot your system every night is - with all respect - a very bad behavior pattern - If your (home) server can't run longer then 24h there is a issue you __must fix__. This is the root of your problem - not the Integration.

meks007, 2025-08-17:
> This HomeAssistent Integration have the **requirement** of an running evcc server instance to be able to start & work [mainly cause during startup the evcc configuration is required to configure the right amount of entities per loadpoint and other dynamic stuff].

Hi,

instead of hitting on your community you should think about the startup logic you built. Understand that evcc exists as a HA addon and that on HAOS, you do not have the possibility to define dependencies. Now, if you provide an integration that is supposed to work upon the evcc HA addon, you *HAVE TO BE ABLE TO* handle your start of the integration correctly.

Implement a delay timer in your integration that continuously checks if evcc is available and waits, if its not.

Don't blame your users that they're doing anything wrong, when in fact they can't do anything about it.

marq24, 2025-08-17:
> ... if you provide an integration that is supposed to work upon the evcc HA addon...

Let me know where you found this info... 

implement it yourself - create a PR

meks007, 2025-08-17:
Sure thing, right on your repo:

[image]

meks007, 2025-08-17:
Now the point is, this issue is incorrectly labelled as invalid. Remove the label, leave the issue open and acknowledge this is a shortcoming of the integration. That way you open up for the community to get creative. By hiding away issues you have all you demonstrate is ignorance.

marq24, 2025-08-17:
Mach einfach einen PR - der Vollumfänglich funktioniert - inlc. Deaktivierung/Reaktivierung der Integration incl der Services, mit WebSocket und ohne... 

ansonsten wünsch ich einen schönen Sonntag

PR #168 (2025-08-17): Retry initialization if evcc is not available
Attempt to fix #158 and #160