marq24/ha-evcc #161: Deprecated function platform_translations has to be replaced by platform_data.platform_translations until HA core 2026.8

Organized92, 2025-08-07:
Since I updated HA core to 2025.8 I'm getting this message in the logs:

Logger: homeassistant.helpers.entity_platform
Quelle: helpers/deprecation.py:222
Erstmals aufgetreten: 07:10:10 (1 Vorkommnis)
Zuletzt protokolliert: 07:10:10

The deprecated function platform_translations was called from evcc_intg. It will be removed in HA Core 2026.8. Use platform_data.platform_translations instead, please create a bug report at https://github.com/marq24/ha-evcc/issues

marq24, 2025-08-07:
Dankeschön - mal gucken, ob ich as bis 2026 geregelt bekomme ;-)

Organized92, 2025-08-07:
After updating to 2025.8.0 the message has not appeared again in my logs. Danke! 👍

commit 17f2832a (2025-08-09): added backward compatibility for older HA versions (caused by fix #161)