marq24/ha-evcc #171: documentation uses wrong character-code

tackin, 2025-08-19:
### Checklist

- [x] I confirm that I have [read & understand the main Integration 'check-list'](https://github.com/marq24/ha-evcc/discussions/166)
- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [ ] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainer of this repo will spend time explaining the world to me — that's marq24's job!. | I live by the motto: Better to ask twice, than to think about it once. | It's marq24's own fault that he provides open-source software and is willing to offer free support.

### Add a description

In [doc](https://github.com/marq24/ha-evcc/blob/main/HA_AS_EVCC_SOURCE.md) you provide code snippets for evcc.yaml.
In the code-examples you use all "-" wrong. That causes errors like: 
```
evcc  | [main  ] FATAL 2025/08/19 11:44:57 configfile failed reading config file: While parsing config: yaml: line 3: mapping values are not allowed in this context
evcc  | [main  ] FATAL 2025/08/19 11:44:57 will attempt restart in: 15m0s

```

 You use:
Character 	—
Character name: EM DASH
Hex code point: 2014
Decimal code point: 8212
Hex UTF-8 bytes: E2 80 94
Octal UTF-8 bytes: 342 200 224

You need to use:
Character 	-
Character name: HYPHEN-MINUS
Hex code point: 002D
Decimal code point: 45
Hex UTF-8 bytes: 2D
Octal UTF-8 bytes: 55

### Add your DEBUG log output

```shell
Attaching to evcc
evcc  | [main  ] INFO 2025/08/19 11:45:33 evcc 0.207.3
evcc  | [main  ] INFO 2025/08/19 11:45:33 UI listening at :7070
evcc  | [main  ] FATAL 2025/08/19 11:45:33 configfile failed reading config file: While parsing config: yaml: line 3: mapping values are not allowed in this context
```

marq24, 2025-08-19:
please close when considered as fixed by latest commit

tackin, 2025-08-19:
I don't know if other code-snippets may be effected as well. Please keep an eye on this as this specific error is a real PITA in YAML.

closed 2025-08-19 by tackin