marq24/ha-evcc #176: Integration unavailable after system restart

thecem, 2025-09-06:
### Checklist

- [x] I confirm that I have [read & understand the main Integration 'check-list'](https://github.com/marq24/ha-evcc/discussions/166)
- [x] I confirm, that I my evcc instance is configured, is up & running and that **I can access the default evcc webinterface**
- [x] My home assistant version is up to date.
- [x] I am using the **latest** version of the integration | See the [release list @github](https://github.com/marq24/ha-evcc/releases) or in home assistant use the HACS 'Update Information' function to ensure that the **latest** released version of the integration is installed. This HACS function can be accessed via the 3-dot menu on the right side for each integration in the HACS integration list.
- [x] I have [checked all issues (incl. the closed ones)](https://github.com/marq24/ha-evcc/issues?q=is%3Aissue+is%3Aclosed) for similar issues in order to avoid reporting a duplicate issue .
- [x] I have prepared DEBUG log output (for technical issues) | In most of the cases of a technical error/issue I would have the need to ask for DEBUG log output of the integration. There is a short [tutorial/guide 'How to provide DEBUG log' here](https://github.com/marq24/ha-senec-v3/blob/master/docs/HA_DEBUG.md)
- [x] I confirm it's really an issue | In the case that you want to understand the functionality of a certain feature/sensor Please be so kind and make use if the discussion feature of this repo (and do not create an issue) - TIA
- [x] I confirm, that I did not read any of the previous bulletin-points and just checked them all. | I don't wanted to waste my time with details, I don't read or follow any existing instructions. | Instead, I want that the maintainer of this repo will spend time explaining the world to me — that's marq24's job!. | I live by the motto: Better to ask twice, than to think about it once. | It's marq24's own fault that he provides open-source software and is willing to offer free support.

### Add a description

After a System restart the integration is not loaded.
A reload or a reconfigure is not solving the problem.
Only a second start of HA solves the problem.

I think there are two problems, once is shown in the log, second is that the reload and reconfigure process is not failsave.

thx!

### Add your DEBUG log output

```shell
Logger: homeassistant.config_entries
Quelle: config_entries.py:751
Erstmals aufgetreten: 08:23:48 (7 Vorkommnisse)
Zuletzt protokolliert: 08:29:35
Error setting up entry evcc for evcc_intg

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 751, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/evcc_intg/__init__.py", line 111, in async_setup_entry
    if not await coordinator.read_evcc_config_on_startup(hass):
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/evcc_intg/__init__.py", line 408, in read_evcc_config_on_startup
    if Version(_version_info) >= Version("0.200.0"):
       ~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/packaging/version.py", line 202, in __init__
    raise InvalidVersion(f"Invalid version: {version!r}")
packaging.version.InvalidVersion: Invalid version: ''
```

thecem, 2025-09-06:
I found the problem exidentially if evcc is the nightly version, it comes to those problem.

marq24, 2025-09-06:
[forgot to press the 'comment' buton]
check your evcc api states output - I guess there is no `version` attribute present... 

... so you found the root issue yourself - normally also nightly build version info should be handled - could be that evcc have change the pattern - I must double check...

thecem, 2025-09-06:
[image] and version in api is "" so this is the problem. Will open an evcc issue.

sry! thx!

closed 2025-09-06 by thecem